bancika/diy-layout-creator #506: Copy paste failed

stinga11, 2020-04-02:
Greetings,
1.When a copy a capacitor or a resistor with a name the copy replace the name with a generic name(C1 or R1)
2. When I open a old project with wires on that project I cant edit the wire options. The only way to modify is insert a new wire and delete the old one.

bancika, 2020-04-02:
1) you can turn off option "Renumber on paste" from Config
2) can you zip and upload a file that cases this problem?

stinga11, 2020-04-02:
[Princeton Reverb Without Reverb and Tremolo layout.zip](https://github.com/bancika/diy-layout-creator/files/4422527/Princeton.Reverb.Without.Reverb.and.Tremolo.layout.zip)
Thanks for answering. The first error was solved by removing that option. The second is still present. There I send you a project.

bancika, 2020-04-02:
Fixed. Will include in the release.

closed 2020-04-02 by bancika