bancika/diy-layout-creator #510: One particular trace cut not recognised by 'highlight connected areas'

timdorrington, 2020-04-04:
I have a layout where an instance of a 'trace cut' on a stripboard conductor is not being recognised when using the 'highlight connected areas' mode and clicking either side of it - the green highlight goes straight through it.  It's only happening for this particular instance - other trace cuts within the same or my other layouts do not have the same problem.  I've tried replacing it with a fresh trace cut or copying & pasting one of the other (working) trace cuts, but the problem persists.

I'm running v4.1.0 on Windows 10.  Log file and layout file are attached.  The offending trace cut is at AH17 on the right-hand board.  Interestingly,  if I introduce another trace cut somewhere above or below it (i.e., on the same conductor strip), the highlight now recognises both trace cuts; if I then remove the new trace cut, the problem returns.

Thanks!

[log-2020-04-04 06.05.57.txt](https://github.com/bancika/diy-layout-creator/files/4432055/log-2020-04-04.06.05.57.txt)

[RAM board.zip](https://github.com/bancika/diy-layout-creator/files/4432061/RAM.board.zip)

bancika, 2020-04-04:
I don't think it's an issue. You have two horizontal red wires running between stripes 12 and 29. One is above, the other is above the cut, so the cut doesn't do anything. Delete one of those wires and it starts working.

[image]

timdorrington, 2020-04-04:
So sorry to not have spotted that.

closed 2020-04-04 by timdorrington