bancika/diy-layout-creator #512: Cannot open app on osx High Sierra 10.13.5

Jahnici, 2020-04-06:
Hi!

I downloaded DIYLC app but it wont open at all. when i double click the DIYLC.app it only shows DIYLC tittle on os x menu bar but it dont open any GUI.

Any Help to get app working ?

bancika, 2020-04-06:
Sierra has issues with the new DIYLC. I suggest staying with version 3.57 for now, as I do not have a solution.

bancika, 2020-04-12:
Actually, I think I just found a workaround. The dedicated OSX build fails on Sierra and El Captain, but you can still run the multi-platform version. Just download diylc-4.3.0.zip (without the "osx" in the name), extract it, change the permissions of run.sh file to be able to run it using

sudo chmod 755 run.sh

and run using

./run.sh

This works on my test machine with Sierra.

closed 2020-04-12 by bancika