bancika/diy-layout-creator #518: Wishlist: bend resistor legs option

dbonin360, 2020-04-09:
Resistor leads cannot bend.  It'd be nice to have the option to "unlock" the leads so they are bendy, like wires?  This would make point to point bends, say  across a cap, or between pins on a tube socket, more natural and visually representative of what we often do with leads.  I'm not sure if other components would benefit from this feature as well.

Love the work you have been doing on this!  Bought you a coffee this week!

Thanks!
Don

bancika, 2020-04-09:
Hi Don,

actually, I've had that request in the queue for almost 10 years now. It's not easy to do and another layer of complexity is that it needs to be done to be backward compatible with all the existing drawings that use resistors....but I'll give it another try. One good thing about the quarantine is that I have more time for the hobbies like DIYLC.

And thanks for the coffee. I got a bean bag :)

All the best,
Bane

EL34xyz, 2020-04-12:
It would be nice to be able to bend the leads on components
I see the complexity of doing it.

Just throwing out ideas - see the attached image

If I shrink a resistor down to just the body and then attach a wire, it looks a bit funky
Like the wire is laying on top of the resistor
If I bring the resistor forward, you can still see the nub of the lead on top of the wire

Don't know if this is possible, 
An edit option for automatically adding a wire to the component
The body would only have attachment points, no nub sticking out

The wire color should be the same color and size as the stock lead color so you don't have to do a bunch of wire editing.

I know many people would like a faster and better way to add leads that can be bent on components

[image]

bancika, 2020-04-17:
I'll look into automating that process. Then it could be extended to transistors and other components.

bancika, 2020-04-18:
Been working on it the last couple of days and I think I have a usable solution. There's a menu option now that replaces fixed legs on leaded components with flexible ones, but also adds flexible leads to components that need them to connect to other components, like tube sockets

[image]

the result is this

[image]

each lead is a separate hookup wire that is modeled to look like a lead and can be edited separately.

And here's what it does to transistors, tube sockets and pots

[image]

EL34xyz, 2020-04-18:
Awesome!
Is there a way to set the number of wire segments as a default?
For example, I like to use 3, 4 or 5 segment to get a nice smooth wire bend

bancika, 2020-04-18:
Whatever default number of control points you set for the wire will be
honored here as well.

EL34xyz, 2020-04-18:
Nice!
That's a great feature you are working on
Thanks Bane

dbonin360, 2020-04-18:
Very nice!

closed 2020-04-19 by bancika