bancika/diy-layout-creator #520: I have a suggestion

stinga11, 2020-04-11:
Greetings, is it possible that I add these types of rectification tube (5Y3) (GZ34) to the schematic drawings?

https://www.thetubestore.com/lib/thetubestore/schematics/Fender/Fender-64-Vibroverb-Custom-Schematic.pdf

https://www.thetubestore.com/lib/thetubestore/schematics/Fender/Fender-57-Deluxe-Schematic.pdf

stinga11, 2020-04-11:
I try to draw them and so I use them. Here I leave you my attempts.
[Rectifier Tubes.zip](https://github.com/bancika/diy-layout-creator/files/4465109/Rectifier.Tubes.zip)

EL34xyz, 2020-04-12:
This might help Bancika
Pins 2, 4 , 6 and 8 are the only pins used on a standard 8 pin tube socket for all the common tube rectifiers.

[image]

bancika, 2020-04-13:
Added duo-diode symbol

[image]

closed 2020-04-13 by bancika