bancika/diy-layout-creator #521: OSX El Captain issue

MacFly77, 2020-04-12:
I can not open the latest version oF DIYLC (4.3) on my older (early 2009) iMac running El Captain 10.11.6.

The "new" icon becomes active...but stops after a few seconds.
New programs are allowed to open.
opening the appletlauncer in the packet contents give following result:

> Last login: Sun Apr 12 10:06:44 on console
> iMac-van-user:~ user$ /Applications/DIYLC.app/Contents/MacOS/JavaAppLauncher ; exit;
> com.apple.macos.useScreenMenuBar has been deprecated. Please switch to apple.laf.useScreenMenuBar
> Exception in thread "main" java.lang.NoSuchFieldError: Splash
> 	at org.diylc.DIYLCSplash.getSplash(Unknown Source)
> 	at org.diylc.DIYLCSplash.<init>(Unknown Source)
> 	at org.diylc.DIYLCStarter.main(Unknown Source)
> logout
> Saving session...
> ...copying shared history...
> ...saving history...truncating history files...
> ...completed.
> 
> [Proces voltooid]
>

bancika, 2020-04-12:
I still don't have a solution for this. Seems like there's a bug with older macs and Java 8...so Java 8 upgrade fixed issues with newer macs, causing issues with older macs. I suggest staying with DIYLC 3.57 until I figure this out.

MacFly77, 2020-04-12:
No problem! as soon as there is an iMac update i'll get a new iMac...

bancika, 2020-04-12:
Actually, I think I just found a workaround. The dedicated OSX build fails on Sierra and El Captain, but you can still run the multi-platform version. Just download diylc-4.3.0.zip (without the "osx" in the name), extract it, change the permissions of run.sh file to be able to run it using

sudo chmod 755 run.sh

and run using

./run.sh

This works on my test machine with Sierra.

closed 2020-04-12 by bancika

MacFly77, 2020-04-12:
Sorry not on El Capitain...

> Exception in thread "main" java.lang.NoClassDefFoundError: org/diylc/DIYLCStarter
> Caused by: java.lang.ClassNotFoundException: org.diylc.DIYLCStarter
> 	at java.net.URLClassLoader$1.run(URLClassLoader.java:202)
> 	at java.security.AccessController.doPrivileged(Native Method)
> 	at java.net.URLClassLoader.findClass(URLClassLoader.java:190)
> 	at java.lang.ClassLoader.loadClass(ClassLoader.java:306)
> 	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:301)
> 	at java.lang.ClassLoader.loadClass(ClassLoader.java:247)

MacFly77, 2020-04-12:
But.....using the zip file without "OSX" and opening the .JAR file (DIYLCStarter) It does work.....4.3 is running!!!

bancika, 2020-04-12:
Great! I'll update the user manual with this info