bancika/diy-layout-creator #523: Stalls/ doesn't finish loading past "DIYLC" splash screen.

BiffDebris, 2020-04-20:
Hello Bancika!
I am using the latest version of DIYLC and have the most current Java JRE, and both x86 and 64-bit I am runnning the latest version of Windows 10 Home Edition. I noticed there is a difference in folderfiles between the exe. download and the .zip download. I have tried both and all it does is get to the opening splash.png and no further. I used a previous version a few months ago and diddn't have a problem except for missing library components that weren't given. Do I have to have a previous version installed before installing the latest or what? I'm not a programmer or have the ability to code, so if it's something obvious to you and other github-ites, my sincerest apologies.
Thank You very much for looking into this.
[log-2020-04-20 02.03.41.txt](https://github.com/bancika/diy-layout-creator/files/4505398/log-2020-04-20.02.03.41.txt)
[log-2020-04-20 02.03.32.txt](https://github.com/bancika/diy-layout-creator/files/4505406/log-2020-04-20.02.03.32.txt)

Thank you for reporting issues and contributing ideas and features to DIYLC repository!

Before you proceed, please read the following:

* If you are reporting a bug, please attach the latest application log file together with description of steps that lead to the issue. They can be found in the user directory under `.diylc/logs` directory. On Windows 7 and later, it's located in `c:/Users/<your username>/.diylc/logs`. On Linux and MacOS, it's typically in `/Users/<your username>/.diylc/logs`.
* If you are having issues starting the application under MacOS, make sure that 3rd party apps are allowed on your Mac. This document covers how to do it - https://support.apple.com/en-us/HT202491.
* Try to look for similar unresolved issues that are already reported to avoid duplicates.
* Make sure that you are running the latest version of DIYLC.

bancika, 2020-04-20:
Can you please try the previous version?

BiffDebris, 2020-04-22:
Hello again;
I couldn't remember which version it was, so I started with the earliest one I could find (3.28.0) and that version works. 4.4.0 wouldn't work. Same problem, the intro splash screen comes up and nothing else happens. I decided then to run DISM, then System File Check and found no corruptions.  I then installed the current version and had same problem. I I tried the .exe install file and the zip file to install. I run W-10 Home Edition and the .diylc log files are attached. I just don't get it. You have a good to great program because IMHO the others don't fulfill the needs of guitarists/musicians or folks that like to tinker and build. I/we need regular pots on our layout, not necessarily trimmers. Same goes for a jacks, types of switches, etc. Most other programs fit the needs of general electronic engineers. And your program is easier for a self-taught novice, admittedly ignorant techie when it comes to electronics. I was bad at math and all the courses in school first hit you with and focus on the math part of the cirriculum. They should have the math AND show you hands-on experience with components and basic circuits. You know, the _fun_ stuff!
[log-2020-04-22 01.35.08.txt](https://github.com/bancika/diy-layout-creator/files/4518425/log-2020-04-22.01.35.08.txt)
[log-2020-04-22 01.26.57.txt](https://github.com/bancika/diy-layout-creator/files/4518426/log-2020-04-22.01.26.57.txt)
[log-2020-04-22 01.12.23.txt](https://github.com/bancika/diy-layout-creator/files/4518427/log-2020-04-22.01.12.23.txt)

closed 2020-04-22 by BiffDebris