bancika/diy-layout-creator #524: Resistor leads are not showing when tightly bent

HelixOnGit, 2020-04-21:
When inserting a resistor (207 shape) across two rows (most thight bend of leads) the leads are not displayed. 
[image]

Thank you for reporting issues and contributing ideas and features to DIYLC repository!

Before you proceed, please read the following:

* If you are reporting a bug, please attach the latest application log file together with description of steps that lead to the issue. They can be found in the user directory under `.diylc/logs` directory. On Windows 7 and later, it's located in `c:/Users/<your username>/.diylc/logs`. On Linux and MacOS, it's typically in `/Users/<your username>/.diylc/logs`.
* If you are having issues starting the application under MacOS, make sure that 3rd party apps are allowed on your Mac. This document covers how to do it - https://support.apple.com/en-us/HT202491.
* Try to look for similar unresolved issues that are already reported to avoid duplicates.
* Make sure that you are running the latest version of DIYLC.

HelixOnGit, 2020-04-21:
[log-2020-04-21 10.24.14.txt](https://github.com/bancika/diy-layout-creator/files/4512535/log-2020-04-21.10.24.14.txt)

bancika, 2020-04-22:
will fix

closed 2020-04-22 by bancika