bancika/diy-layout-creator #525: Wishlist: Rotary Switch

HelixOnGit, 2020-04-21:
Please include higher number of throw and different numbers of poles (3P4T, 1P12T, 1PxT etc.) for the rotary switch

bancika, 2020-04-22:
The rotary switch I added in this iteration is the large one used in amp building. Those have only few positions. This would need a new component type of component or maybe even two.

[image]
[image]

HelixOnGit, 2020-04-22:
I think these types are pretty much the most popular in synth diy community (quite similar to your first picture)
https://lorlinelectronics.co.uk/catalogue?filter_switch-category=rotary-switches
They also come as solder lug/pcb mount terminals. 

I understand that you maybe aim at a different target audience though.

bancika, 2022-10-23:
These were finally added in 4.30, 4 types of sealed and 4 types of open switches.

closed 2022-10-23 by bancika