bancika/diy-layout-creator #526: V4.5.0 Film Capacitor (Radial) now needs >= 4 rows to lay flat

chipshorter, 2020-04-23:
As per title, the Film Cap (Radial) is "on-end" unless stretched across >= 4 rows in V4.5.0

It was happy with 3 rows in V4.3.0

bancika, 2020-04-23:
Not sure I follow, do you have a screenshot?

animalejo, 2020-04-23:
i join to this

Video: https://drive.google.com/file/d/1F_8whykhgVCQxo0nV_o0SYxTjaL-RI9l/view?usp=sharing

bancika, 2020-04-23:
I see. Will fix

animalejo, 2020-04-23:
the same with the resistors, the leads disapears at the 4rows, with real dimention, 7mm length

chipshorter, 2020-04-23:
> i join to this
> 
> Video: https://drive.google.com/file/d/1F_8whykhgVCQxo0nV_o0SYxTjaL-RI9l/view?

> Not sure I follow, do you have a screenshot?

Sorry, I should have made my report clearer. But thanks to animalejo for the video I think you have it.

bancika, 2020-04-24:
Fixed. WIll go out with the next version

closed 2020-04-24 by bancika