bancika/diy-layout-creator #527: Breadboard selection

ultrasteve54, 2020-04-23:
When you select breadboard from dropdown list and position in layout area, left click to select other components the breadboard  is non responsive to any selection and layout screen is frozen until you undo select breadboard from edit menu. Veroboard layout works ok.
[log-2020-04-23 03.11.10.txt](https://github.com/bancika/diy-layout-creator/files/4523060/log-2020-04-23.03.11.10.txt)

bancika, 2020-04-23:
Can you please post your configuration file? I couldn't reproduce with default configuration.

ultrasteve54, 2020-04-23:
Config file as requested
[config.xml.txt](https://github.com/bancika/diy-layout-creator/files/4523847/config.xml.txt)

closed 2020-04-23 by ultrasteve54

bancika, 2020-04-24:
I can reproduce it now. It's the orientation of the board that triggers the
bug. Will fix