bancika/diy-layout-creator #529: Very slow / unusable on OSX Mojave

flpvsk, 2020-04-26:
Not sure if it's a configuration issue or something else, but the program is slow to the point of being unusable on OSX Mojave.

* Whenever I type anything in the tree search, letters appear with a delay and UI hangs, I can't do anything else;
* When I drag components in the UI, the refresh rate is very low;
* Eventually the app crashes;

Please see video: [link](https://drive.google.com/file/d/15DHZXryip4gkkiYuKmE6sS0fwI5gNwtJ/view?usp=sharing)

Java version: 1.8.0_251-b08
DIYLC version: Version 4.5.0 (1.0)

System info: 
[image]
[image]

htor, 2020-04-27:
i'm also seeing slow tree search on v4.5.0 mac os mojave, although no crashes so far or unusable UI.

selecting Config > Toolbox > Tabbed toolbar instead of Searchable Tree makes things faster for me.

maybe try fiddling with the hardware acceleration and hi-quality settings under Config will help

flpvsk, 2020-04-27:
@htor thanks, "Config > Toolbox > Tabbed toolbar" seems to have fixed the issue 🎩

bancika, 2020-04-27:
Let's leave it open for now, the tree should work fine

bancika, 2020-05-04:
Can you please try this prototype version www.diy-fever.com/temp/diylc-4.6.0-osx.zip
I made an experimental change that could make it better. Seems like there's a bug in Java introduced for OSX that makes this performance issue.
Cheers

flpvsk, 2020-05-04:
Just tried – the issue is still there

bancika, 2020-09-17:
Can you please try this version and see if it helps?
[diylc-4.9.0-osx.zip](https://github.com/bancika/diy-layout-creator/files/5240254/diylc-4.9.0-osx.zip)
[diylc-4.9.0.zip](https://github.com/bancika/diy-layout-creator/files/5240255/diylc-4.9.0.zip)

flpvsk, 2020-09-22:
@bancika seems like it's fixed in that release, thank you!

bancika, 2020-09-22:
awesome!

closed 2020-09-22 by bancika