bancika/diy-layout-creator #532: Can't rotate Tag strip : (Popup :Some components cannot be rotated)

PH0MB, 2020-04-28:
Thank you for reporting issues and contributing ideas and features to DIYLC repository!

Before you proceed, please read the following:

* If you are reporting a bug, please attach the latest application log file together with description of steps that lead to the issue. They can be found in the user directory under `.diylc/logs` directory. On Windows 7 and later, it's located in `c:/Users/<your username>/.diylc/logs`. On Linux and MacOS, it's typically in `/Users/<your username>/.diylc/logs`.
* If you are having issues starting the application under MacOS, make sure that 3rd party apps are allowed on your Mac. This document covers how to do it - https://support.apple.com/en-us/HT202491.
* Try to look for similar unresolved issues that are already reported to avoid duplicates.
* Make sure that you are running the latest version of DIYLC.

bancika, 2020-05-04:
for the time being, you can rotate them by changing orientation through the edit dialog...just double click on a component

PH0MB, 2020-05-11:
That did the job! tnx

bancika, 2020-12-11:
implemented for 4.12

closed 2020-12-11 by bancika