bancika/diy-layout-creator #533: Dragging issues

Delwick, 2020-04-30:
Hi

I'm a new user and have 4.5.0 downloaded on my MAC.

This app seem perfect for my needs, however I cannot drag with my mouse, whether it be to select multiple items, or to actually move an item.

Also double clicking does not select an item.

Is this simply operator error please?

Thank you for reporting issues and contributing ideas and features to DIYLC repository!

Before you proceed, please read the following:

* If you are reporting a bug, please attach the latest application log file together with description of steps that lead to the issue. They can be found in the user directory under `.diylc/logs` directory. On Windows 7 and later, it's located in `c:/Users/<your username>/.diylc/logs`. On Linux and MacOS, it's typically in `/Users/<your username>/.diylc/logs`.
* If you are having issues starting the application under MacOS, make sure that 3rd party apps are allowed on your Mac. This document covers how to do it - https://support.apple.com/en-us/HT202491.
* Try to look for similar unresolved issues that are already reported to avoid duplicates.
* Make sure that you are running the latest version of DIYLC.

thestoneage, 2020-05-22:
Got the same issues running version 4.7.0 on 10.15 with the latest Java 8 Runtime. The curser does not change to the finger when hovering over a component.

EL34xyz, 2020-05-23:
Works for me on 4.7.0
Selecting area and then move the cursor over the selected area until the finger displays, then left click to drag the selected area.

Another way that I like even better is to select an area and use the arrow keys to move one grid unit at a time

Delwick, 2020-05-23:
Guys, sorry for the delay in responding.

It was user mis-operation! As soon as I unchecked 'Highlight connected areas' in the config menu, it all worked fine..

Great program, thanks!

closed 2020-06-05 by bancika