bancika/diy-layout-creator #535: Any Version over 4 will not start on my Mac

FastEdTex, 2020-05-02:
I can't get any version 4.0 or higher to run on my Mac. Java 8 build 221 on OS 10.10.5 I'm running 3.57.0 DIYLC. I don't get any error, it just pauses for 2 seconds and goes back to the finder, no message.
[log-2020-04-29 01.54.19.txt](https://github.com/bancika/diy-layout-creator/files/4566771/log-2020-04-29.01.54.19.txt)

FastEdTex, 2020-05-02:
This is what I get when trying to run the platform generic version.
edward-underwoods-imac-9:diylc-450 FastEd$ ./run.sh
Exception in thread "main" java.lang.UnsupportedClassVersionError: org/diylc/DIYLCStarter : Unsupported major.minor version 52.0
	at java.lang.ClassLoader.defineClass1(Native Method)
	at java.lang.ClassLoader.defineClassCond(ClassLoader.java:637)
	at java.lang.ClassLoader.defineClass(ClassLoader.java:621)
	at java.security.SecureClassLoader.defineClass(SecureClassLoader.java:141)
	at java.net.URLClassLoader.defineClass(URLClassLoader.java:283)
	at java.net.URLClassLoader.access$000(URLClassLoader.java:58)
	at java.net.URLClassLoader$1.run(URLClassLoader.java:197)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.net.URLClassLoader.findClass(URLClassLoader.java:190)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:306)
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:301)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:247)

bancika, 2020-05-02:
seems like you are running java6. You need java8 for diylc 4 and later

FastEdTex, 2020-05-02:
System Preferences Java controll panel shows I’m running 8

bancika, 2020-05-02:
try typing "java -version" in the terminal

FastEdTex, 2020-05-02:
Now that’s a bitch. Controls panel report runtime 8

edward-underwoods-imac-9:~ FastEd$ java -version
java version "1.6.0_65"
Java(TM) SE Runtime Environment (build 1.6.0_65-b14-468-11M4833)
Java HotSpot(TM) 64-Bit Server VM (build 20.65-b04-468, mixed mode)

> On May 2, 2020, at 12:43 PM, Bane Stojkovic <notifications@github.com> wrote:
> 
> java -version

bancika, 2020-05-02:
Here's how to change default java version

https://intellipaat.com/community/8277/how-to-set-or-change-the-default-java-jdk-version-on-os-x

FastEdTex, 2020-05-02:
followed that and it reports I have version 6.
I goto controll panel/java and it shows 8.

Evidently it (8) is in a different directory.

bancika, 2020-05-02:
the steps show how to change it to 8

FastEdTex, 2020-05-02:
Well I don’t think so.
every time I type “java -version” it reports version 6.
So I’m not sure how setting the JAVA_HOME is going to help.

Last login: Sat May  2 13:31:38 on ttys000
edward-underwoods-imac-9:~ FastEd$ which java
/usr/bin/java
edward-underwoods-imac-9:~ FastEd$ java -version
java version "1.6.0_65"
Java(TM) SE Runtime Environment (build 1.6.0_65-b14-468-11M4833)
Java HotSpot(TM) 64-Bit Server VM (build 20.65-b04-468, mixed mode)
edward-underwoods-imac-9:~ FastEd$ 
edward-underwoods-imac-9:~ FastEd$ JAVA_HOME=/Library/Java/Home
edward-underwoods-imac-9:~ FastEd$ $JAVA_HOME
-bash: /Library/Java/Home: is a directory
edward-underwoods-imac-9:~ FastEd$

bancika, 2020-05-02:
java -version checks the current version, does not set it to anything.

setting JAVA_HOME actually changes the default version of java and that's
what you need.

FastEdTex, 2020-05-02:
Is that what I did? See red below. Looks like it worked.
But this is what I get:
edward-underwoods-imac-9:Library FastEd$ java -version
java version "1.6.0_65"
Java(TM) SE Runtime Environment (build 1.6.0_65-b14-468-11M4833)
Java HotSpot(TM) 64-Bit Server VM (build 20.65-b04-468, mixed mode)
edward-underwoods-imac-9:Library FastEd$

FastEdTex, 2020-05-02:
> notifications@github.com <mailto:notifications@github.com>>

I give up.

I can’t get Java 8 to work.
I have installed the latest update 4 time and it shows that in the control panel “Java”
but from the terminal command it shows version 6.

If I remove version 6 and install again Java 8, then DIYLC 57 will not run or any other version of DIYLC.
If I re-install Java 6 then at least I can get 57 to run.

Ed

bancika, 2020-05-02:
Did you do anything other than typing "java -version", because you didn't
send anything else in any of the messages

FastEdTex, 2020-05-02:
Bane,

This is what i typed: computer prompts in black, my typing in red, computer echo in Green

edward-underwoods-imac-9:~ FastEd$ which java
/usr/bin/java

edward-underwoods-imac-9:~ FastEd$ java -version
ava version “1.6.0_65"
Java(TM) SE Runtime Environment (build 1.6.0_65-b14-468-11M4833)
Java HotSpot(TM) 64-Bit Server VM (build 20.65-b04-468, mixed mode)

edward-underwoods-imac-9:~ FastEd$ JAVA_HOME=/Library/Java/Home

edward-underwoods-imac-9:~ FastEd$ $JAVA_HOME
-bash: /Library/Java/Home: is a directory

Ed

FastEdTex, 2020-05-03:
don’t know what this means, don’t understand the path but it shows JRE

edward-underwoods-imac-9:Library FastEd$ /Library/Internet\ Plug-Ins/JavaAppletPlugin.plugin/Contents/Home/bin/java -version
java version "1.8.0_251"
Java(TM) SE Runtime Environment (build 1.8.0_251-b08)
Java HotSpot(TM) 64-Bit Server VM (build 25.251-b08, mixed mode)

bancika, 2020-05-03:
this should set it to 1.8

export JAVA_HOME=`/usr/libexec/java_home -v 1.8`

FastEdTex, 2020-05-03:
edward-underwoods-imac-9:~ FastEd$ export JAVA_HOME=`/usr/libexec/java_home -v 1.8`
Unable to find any JVMs matching version "1.8".
edward-underwoods-imac-9:~ FastEd$ 

> On May 3, 2020, at 2:13 AM, Bane Stojkovic <notifications@github.com> wrote:
> 
> export JAVA_HOME=`/usr/libexec/java_home -v 1.8`

bancika, 2020-05-03:
Bummer. In some versions of OSX it's different. Maybe this would work
https://medium.com/@bryantjiminson/changing-default-java-version-in-mac-os-x-yosemite-470f11a6084b

FastEdTex, 2020-05-03:
edward-underwoods-imac-9:libexec FastEd$ ./java_home -V
Matching Java Virtual Machines (3):
    14.0.1, x86_64:	"Java SE 14.0.1"	/Library/Java/JavaVirtualMachines/jdk-14.0.1.jdk/Contents/Home
    1.6.0_65-b14-468, x86_64:	"Java SE 6"	/Library/Java/JavaVirtualMachines/1.6.0.jdk/Contents/Home
    1.6.0_65-b14-468, i386:	"Java SE 6"	/Library/Java/JavaVirtualMachines/1.6.0.jdk/Contents/Home

/Library/Java/JavaVirtualMachines/jdk-14.0.1.jdk/Contents/Home

bancika, 2020-05-03:
You don't seem to have java 8 there

FastEdTex, 2020-05-04:
These are screen capture of the control panel for Java

I got this terminal command from here https://www.java.com/en/download/help/version_manual.xml#cmdline <https://www.java.com/en/download/help/version_manual.xml#cmdline>

edward-underwoods-imac-9:libexec FastEd$ /Library/Internet\ Plug-Ins/JavaAppletPlugin.plugin/Contents/Home/bin/java -version
java version "1.8.0_251"
Java(TM) SE Runtime Environment (build 1.8.0_251-b08)
Java HotSpot(TM) 64-Bit Server VM (build 25.251-b08, mixed mode)

bancika, 2020-05-04:
This directory should contain all installed JREs, what do you see there?

/Library/Java/JavaVirtualMachines

FastEdTex, 2020-05-04:
edward-underwoods-imac-9:~ FastEd$ cd /Library/Java/JavaVirtualMachines
edward-underwoods-imac-9:JavaVirtualMachines FastEd$ ls
1.6.0.jdk	jdk-14.0.1.jdk

bancika, 2020-05-04:
yeah, JRE 8 doesn't seem to be installed. You can grab it here
https://mac.filehorse.com/download-java-runtime/10671/

FastEdTex, 2020-05-04:
Bane,

I’ve come to the conclusion that my Mac hardware is not capable of running Java 8.
I have an older Mac - iMac Early 2009
I think this may be the problem.

Ed

bancika, 2020-05-04:
It shouldn't be the case, since you have Java 14 installed :)

FastEdTex, 2020-05-04:
edward-underwoods-imac-9:~ FastEd$ java -version
Error: dl failure on line 558
Error: failed /Library/Java/JavaVirtualMachines/jdk-14.0.1.jdk/Contents/Home/lib/server/libjvm.dylib, because dlopen(/Library/Java/JavaVirtualMachines/jdk-14.0.1.jdk/Contents/Home/lib/server/libjvm.dylib, 10): no suitable image found.  Did find:
	/Library/Java/JavaVirtualMachines/jdk-14.0.1.jdk/Contents/Home/lib/server/libjvm.dylib: code signature invalid for '/Library/Java/JavaVirtualMachines/jdk-14.0.1.jdk/Contents/Home/lib/server/libjvm.dylib’

FastEdTex, 2020-05-13:
Bane,

Got a new Mac, loaded Java 8 first try.
DIYLC 4.6 working.

It has something to do with the apple version of Java on my Mac. Would not install 8 or let me get rid of 6.

Ed

bancika, 2020-05-13:
Awesome, thanks for the update!

closed 2020-05-16 by bancika