bancika/diy-layout-creator #536: Unable to start v4.5.0 on my MacBook Pro

rexmaxus, 2020-05-02:
How do I get v4.5.0 to open on my Mac? If I click the icon it just goes back to Finder, if I try opening diylc.jar it warns me to use run.sh, but when I use run.sh it just opens a single line in Xcode.

3rd party apps are allowed on my device.

bancika, 2020-05-02:
sh files cannot be double clicked. Just exec it from the console by typing
./run.sh

rexmaxus, 2020-05-02:
typing ./run.sh in Terminal results in "-bash: ./run.sh: No such file or directory"

bancika, 2020-05-16:
did you download the OSX DIYLC package? It should extract DIYLC.app that is executable. Just move it to "Applications"

awasson, 2020-06-06:
I'm experiencing the same issue. I'm on Mac OS X Sierra 10.12.6 with Java 8 installed. Previously I was running a variant of the 3.x version with Legacy Java 6 interpreter installed. I've tried 4.0, 4.5 and 4.7. None work. It fails silently without any message or error. 

I have backed down to 3.57.0 and it runs. As per usual, I ran sudo spctl --master-disable to get it to run the first time but I did not need to reinstall Legacy Java 6.

awasson, 2020-06-07:
Just an FYI. I was able to run the diylc.jar file for v4.7.0 on my Mac with OS X 10.12.6 and it does indeed run. The run.sh does not but the Java Jar file does.

mstratman, 2020-06-10:
> Just an FYI. I was able to run the diylc.jar file for v4.7.0 on my Mac with OS X 10.12.6 and it does indeed run. The run.sh does not but the Java Jar file does.

Did you just do `java -jar diylc.jar` or something different?

I'm also on OS X 10.12.6, and I get the following:

```
java -jar diylc.jar
Error: A JNI error has occurred, please check your installation and try again
Exception in thread "main" java.lang.NoClassDefFoundError: org/diylc/common/IPlugInPort
	at java.lang.Class.getDeclaredMethods0(Native Method)
	at java.lang.Class.privateGetDeclaredMethods(Class.java:2701)
	at java.lang.Class.privateGetMethodRecursive(Class.java:3048)
	at java.lang.Class.getMethod0(Class.java:3018)
	at java.lang.Class.getMethod(Class.java:1784)
	at sun.launcher.LauncherHelper.validateMainClass(LauncherHelper.java:544)
	at sun.launcher.LauncherHelper.checkAndLoadMain(LauncherHelper.java:526)
Caused by: java.lang.ClassNotFoundException: org.diylc.common.IPlugInPort
	at java.net.URLClassLoader.findClass(URLClassLoader.java:381)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:331)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	... 7 more
```

bancika, 2020-11-19:
should be fixed in 4.10.0, but please let me know if it's still an issue.

closed 2020-11-19 by bancika