bancika/diy-layout-creator #538: 4.6.0 refuses to load some previous version projects

Yupnitski, 2020-05-05:
Hi!
I'm having issues with loading some projects that have been created with earlier version of DIYLC. I don't recall with what specific version the failing one's have been created with, but they opened fine with 4.5.0. 
Here's an take from the log file:

18:01:02,759  WARN [main] ProjectFileManager - Could not open with the new xStream, trying the old one
com.thoughtworks.xstream.converters.ConversionException: Invalid reference
---- Debugging information ----
reference           : ../../org.diylc.components.connectivity.TraceCut/fillColor
class               : org.diylc.components.passive.RadialElectrolytic
required-type       : org.diylc.components.passive.RadialElectrolytic
converter-type      : com.thoughtworks.xstream.converters.reflection.ReflectionConverter
path                : /org.diylc.core.Project/components/org.diylc.components.passive.RadialElectrolytic/labelColor
class[1]            : java.util.ArrayList
converter-type[1]   : com.thoughtworks.xstream.converters.collections.CollectionConverter
class[2]            : org.diylc.core.Project
version             : null

Thank you for reporting issues and contributing ideas and features to DIYLC repository!

Before you proceed, please read the following:

* If you are reporting a bug, please attach the latest application log file together with description of steps that lead to the issue. They can be found in the user directory under `.diylc/logs` directory. On Windows 7 and later, it's located in `c:/Users/<your username>/.diylc/logs`. On Linux and MacOS, it's typically in `/Users/<your username>/.diylc/logs`.
* If you are having issues starting the application under MacOS, make sure that 3rd party apps are allowed on your Mac. This document covers how to do it - https://support.apple.com/en-us/HT202491.
* Try to look for similar unresolved issues that are already reported to avoid duplicates.
* Make sure that you are running the latest version of DIYLC.

bancika, 2020-05-05:
Do you have a test file I can use to validate the fix?

Yupnitski, 2020-05-05:
Yeah, sure! See attachment.
This is just one of the many that won't open. As layouts, all of them
should be quite similar in means of what components or setups I've used.

bancika, 2020-05-05:
No attachments :)

I think in order to attach a file you need to use github issues page, not the email. Also, it doesn't accept DIY files, but it does when zipped. Thanks

Yupnitski, 2020-05-05:
Ok! Let's try this way :D

[Sample.zip](https://github.com/bancika/diy-layout-creator/files/4582997/Sample.zip)

bancika, 2020-05-05:
Ok, verified the issue and the fix. It will go out in 4.7.0
Cheers

Yupnitski, 2020-05-05:
Ok, great! THX! :)

boilerkeeper, 2020-05-09:
Hi,
I'm having similar problems opening files created with an older build of diylc.
Here's an excerpt from my log file:

12:41:00,861 ERROR [SwingWorker-pool-1-thread-6] Presenter - Could not load file
com.thoughtworks.xstream.converters.ConversionException: Invalid reference
---- Debugging information ----
reference           : ../../org.diylc.components.connectivity.TraceCut/fillColor
class               : org.diylc.components.semiconductors.DIL_IC
required-type       : org.diylc.components.semiconductors.DIL_IC
converter-type      : com.thoughtworks.xstream.converters.reflection.ReflectionConverter
path                : /org.diylc.core.Project/components/org.diylc.components.semiconductors.DIL_IC/labelColor
class[1]            : java.util.ArrayList
converter-type[1]   : com.thoughtworks.xstream.converters.collections.CollectionConverter
class[2]            : org.diylc.core.Project
version             : null
-------------------------------
Thanks!

bancika, 2020-05-16:
that's the same issue, should work

closed 2020-05-16 by bancika