bancika/diy-layout-creator #543: Adjust font size Property  in Components

StephenCohrs, 2020-05-23:
Cannot change the font size in components. e.g. the 25mm potentiometer default font size is too small and it doesn't seem to be a property of the device. Resetting the project default size works but then some of the smaller components font is too big and overruns the borders. Can they be set differently? 
IMO, the default is good but you need to override in some cases - bigger or smaller.
Is there a workaround?

bancika, 2020-05-23:
No workaround at the moment

bancika, 2020-12-13:
Actually, the workaround is to have the component label empty and then overlay a "Label" component with the desired font size. Thinking about a permanent solution.

bancika, 2020-12-13:
will implement in 4.13

closed 2020-12-13 by bancika