bancika/diy-layout-creator #544: Can't rotate images

EL34xyz, 2020-05-23:
I never tried to rotate an image until today

I have some building block that have small images as part of the building block
The normal parts will all rotate but the image won't

[image]

EL34xyz, 2020-05-26:
Bancika, 
Can you add the ability to rotate images?
My example above is an image of a ring terminal with two dots added where you can attach wires

The dots will rotate but not the image

bancika, 2020-05-27:
Yes it is possible, but with the lock-down over there are no guarantees
when it will be done. I did a ton of DIYLC work recently and felt that I
needed to allocate my free time to other things.

EL34xyz, 2020-05-27:
Ok, thanks

bancika, 2020-12-06:
Implemented for 4.11

closed 2020-12-06 by bancika

EL34xyz, 2020-12-06:
Thanks Bane

EL34xyz, 2020-12-06:
I am not seeing 4.11 in the downloads yet?

bancika, 2020-12-06:
Pending release