bancika/diy-layout-creator #545: Drawingon the backside of the PCB

poesel, 2020-05-23:
Is it possible to draw on the backside of a PCB? 
I looked through the docs but didn't find anything.

poesel, 2020-05-26:
So its not possible?

bancika, 2020-05-27:
What do you mean by that? You can send components to the back...but there
is no support for multilayer pcb if that's what you want.

poesel, 2020-05-27:
How? Maybe it’s obvious but I have missed it. Thanks for help.

bancika, 2020-05-27:
Right click on a component, there should be an option to send to back

poesel, 2020-05-28:
Ah, ok. The option was (and is) greyed out but the keyboard shortcut does work.
Two questions:
Why is it greyed out?
How can I flip the board to see the back side?

bancika, 2020-06-02:
I just tried and it's not grayed out for me.
Is it consistently grayed out? Can you upload the project file that causes issues and the exact place where you clicked.

As for the other question, there's no such feature.

poesel, 2020-06-03:
Hmm, if you can't flip the board what is the point of being able to send parts to the backside where you can't see them any more?

bancika, 2020-06-03:
You can see them when you make the board transparent.

closed 2020-12-13 by bancika