bancika/diy-layout-creator #547: Undo is slow

TomKamphuys, 2020-05-26:
Undo takes seconds to take place.

Windows 10, V4.7.0

TomKamphuys, 2020-05-26:
Changing the zoom factor is also slow.

TomKamphuys, 2020-05-30:
It seems to have something to do with a dotted line. Attached you'll find a reduced example differing only in the absence/presence of a dotted line. The difference in visual feedback on a CTRL-Z is substantial.

[slowFast.zip](https://github.com/bancika/diy-layout-creator/files/4706524/slowFast.zip)

TomKamphuys, 2020-05-30:
It looks like the dotted line is not the only thing. Removing the dotted line from my original file does not help.

bancika, 2020-06-02:
Thanks for the info. Yes, it certainly seems to be the dotted line. I'll investigate

bancika, 2020-12-16:
fixed for 4.13

closed 2020-12-16 by bancika