bancika/diy-layout-creator #548: Load Line Editor cooperation

TomKamphuys, 2020-05-27:
I don't know how to reach you otherwise, so I'll try it this way.

I found the Load Line Editor in DIYLC and found out it was added in 4.6.0 for future use. I really like DIYLC and would like to help. I've written a load line plotter for Octave (Matlab clone) and would like to offer help. I lack the Java skills to implement it though.

If you want you can reach me at tomkamphuys at hotmail dot com.

[image]

bancika, 2020-12-13:
Hey, sorry for not responding sooner, I missed this. 
I'm naturally interested. The thing is, I'm still not sure where I want to take Loadline editor and if people will use it, so I've been investing my time in some core functionalities and fixes that people could benefit from.
Do you have any ideas we can use to build on top of what I have already done for loadlines?

TomKamphuys, 2020-12-14:
No problem bancika, I see the new versions are coming in rapid succession. Thanks for the effort!

Good question about the loadlines. I think (/fear) most guitar amp builders just build whatever Fender, Marshall, etc. used. I've plotted a lot of load lines and in the end I .... just used what, in my case, Marshall used. 
During the summer I learned that for the unbypassed case almost no load line plots can be found, other than in some 1950's document. They are completely different from what you think of regarding load lines. DIYLC could be (one of?) the first to show them.
In the last weeks I Iearned that in the hifi world Constant Current Sources are used quite often with adjustable currents, limiting the use of load line plots. 
You could also argue that you leave the layout/schematic 'world' and enter the (lt)spice world.

I hope I'm not too pessimistic, but I think the question is (still) indeed: What does DIYLC want to give the user?

bancika, 2020-12-15:
well, originally I developed another tool independent of DIYLC that was called DIYAS that had loadline tools where you can create loadline from the tube datasheet, then another tool where you can play with loadlines and calculate bias values...and finally there was a tool where you could put together a circuit. The idea was to make it a real amp simulator that would use loadlines somehow, but I lacked knowledge in that department so I gave up on that idea.

in the context of DIYLC, my original idea was to have loadline editor that can be used to create tube loadlines. Then we can insert them into layouts and schematics and set the bias point there, to accompany the schematic or layout. But not sure if that would be too useful.

blackcorvo, 2020-12-22:
I know exactly who'd love this tool: Hi-Fi guys! They're kinda obsessed with specific loadlines, harmonic response, etc.
Guitar is much more forgiving in those regards (you can "fix it in post" so to speak, by choosing a speaker that compliments your amp the way you prefer).

Maybe get some people talking about that feature in places like the DIY Audio forums? They'd dig an easier way to plot lines than drawing them in Paint, I'm sure.

pmfawcett, 2021-01-02:
To do loadlines with full generality, you need good tube models and a back end that is more-or-less a full SPICE implementation.  I have an LTSpice push-pull power tube load line calculator that I developed a number of years ago and which works quite well.  Of course the results are only as good as the tube models, but I have used it to design power amps for several projects.   Probably this is beyond the scope of what naturally would fit in DIYLC.   I have seen a website recently that seems to do something similar on the back, then feed the results to a web interface. It's numbers don't match with my own calculations however, so I would not regard it as reliable.

On the other hand, there are some good quick-and-dirty loadline calculators for the most common guitar preamp tubes that work well and would be easily adapted.  See Merlin's excel-based loadline calculator for example, which you can find at the bottom of his [downloads page](http://www.valvewizard.co.uk/links.html). As all of the calculations are readily visible, this would translate very easily into a program.