bancika/diy-layout-creator #549: Stuck on splash screen (Ubuntu)

neilmb, 2020-05-29:
On Ubuntu, I'm trying to upgrade to version 4.7.0 from 3.57. When I run the `./run.sh` script, I see the splash screen but never get the main window. The log file is here but I don't see any obvious errors. It just gets to the checking on updates and then stops doing anything.

[log-2020-05-29 12.49.16.txt](https://github.com/bancika/diy-layout-creator/files/4703420/log-2020-05-29.12.49.16.txt)

analoganarchist, 2020-05-31:
I have the same problem, but on Win 10. I had version 2.8 I think and went to upgrade, installed it without uninstalling the previous version. I ran it and it just got stuck on the splash screen. Tried reinstalling multiple times, different versions but none of them worked. I tried re-installing the old version and now even that version won't function properly. It gets past the splash screen, but then freezes once the program is open. I really hope there's a fix soon. DIYLC is an important tool for me.

bancika, 2020-06-01:
can you upload your config file?

bancika, 2020-06-01:
also, try temporarily moving the config file from .diylc folder to
somewhere else. The app will start with default configuration and create a
new file. If it works, great.

analoganarchist, 2020-06-01:
Hey, Bane,

Man, it was THAT simple. Having the program create another config file
started it right up and it functions properly! Just in time for a few
projects I'm about to work on.

I'll go upload the old config file right now.

Thanks very much! For the help and DIYLC itself.

analoganarchist, 2020-06-01:
THe old config file:

[configtxtversion.txt](https://github.com/bancika/diy-layout-creator/files/4710602/configtxtversion.txt)

bancika, 2020-06-01:
Yeah, it has to do with theme format.

bancika, 2020-06-02:
I patched the app so it doesn't fail catastrophically in this case. In the meantime, it's best to just rename config file and start over.

closed 2020-06-02 by bancika