bancika/diy-layout-creator #551: Feature request: absolute positioning

guntesundell, 2020-06-01:
Thanks for a great product!

I have a feature request. If you want to position a component exactly, right now the only way is to "nudge" the component using a relative distance from an origin. It would be useful if you could specify exact coordinates. I.e. "move to x,y".

EL34xyz, 2020-06-02:
Sounds tricky to program?
What point in the component is the point you want to put at X and Y?

guntesundell, 2020-06-02:
The same point that is being used for nudge.
Basically, it's just setting the coordinates on the "canvas". Have no idea if it's tricky to program - I haven't coded in 20 years! ;-)

bancika, 2020-06-02:
Yeah, but the component has its size, it's not a singular point. When you
say put component to (x, y)...what does it mean? Should (x, y) land on the
center of component, or top-left corner, or something else?

guntesundell, 2020-06-02:
The idea I had was to use the same point as the one being used for "nudge" (i.e. relative move). In the components I have used (so far) is on the top left corner but I don't know if this differs between components.

EL34xyz, 2020-06-02:
How do you know what the nudge points coords are?
You would have to know that so that you can enter X and Y relative to those coords.
Right?

guntesundell, 2020-06-02:
What I have done so far is to sort of eyeball it and put the component in the upper left corner. Then I have used nudge. It works, but it's awkward...
Don't get me wrong, this a wonderful piece of software - much better fit for purpose than Visio that I have used for layout until now.
A wonderful thing would of cpurse be if you can set the "point" for the component: upper left, upper mid, upper right, center left... and so on. But I don't want to ask for too much ;-)

bancika, 2020-06-02:
nudge is relative, moving 10mm to the right is the same regardless of the
shape of a component. But placing the component on to 10th milimeter could
mean anything

guntesundell, 2020-06-02:
Yes, I understand that. But it all comes down to how the drawing of the component has been implemented - and I don't know that. I would be happy to if I could position the center of the component on a specific x, y point on the canvas. And I would be equally happy if I could use the upper left corner to do the same.