bancika/diy-layout-creator #552: PerfBoard spacing incorrect

TomKamphuys, 2020-06-07:
The PerfBoard spacing is incorrect. In the image below I've set the grid spacing to 3/8".

[image]

The grid lines do not match the holes in the PerfBoard. I've looked at the code and think it's here:
```
    if (componentState != ComponentState.DRAGGING) {
      if (alpha < MAX_ALPHA) {
        g2d.setComposite(AlphaComposite.getInstance(AlphaComposite.SRC_OVER, 1f * alpha / MAX_ALPHA));
      }
      Point p = new Point(firstPoint);
      int holeDiameter = getClosestOdd((int) HOLE_SIZE.convertToPixels());
      int spacing = (int) this.spacing.convertToPixels();

      while (p.y < finalSecondPoint.y - spacing) {
        p.x = firstPoint.x;
        p.y += spacing;
        while (p.x < finalSecondPoint.x - spacing - holeDiameter) {
          p.x += spacing;
          g2d.setColor(Constants.CANVAS_COLOR);
          g2d.fillOval(p.x - holeDiameter / 2, p.y - holeDiameter / 2, holeDiameter, holeDiameter);
          g2d.setColor(borderColor);
          g2d.drawOval(p.x - holeDiameter / 2, p.y - holeDiameter / 2, holeDiameter, holeDiameter);
        }
      }
      super.drawCoordinates(g2d, spacing, project);
    }
  }
```
First the _SPACING_ is copied to _this.spacing_ first (not in this code). Here a local _spacing_ in pixels is calculated from that. In the while loop, it is continuously added to the x and y coordinates. The problem, I think, is that the _convertToPixels_ round it to an integer. You then sum this rounded number, increasing the round off error each time. I would recommend to delay the _convertToPixels_ call as long as possible, to keep the round off error to a minimum.

bancika, 2020-06-08:
good spot

ademenev, 2020-09-03:
In fact, **everything** should be in internal **double** units, and conversion to int performed just before drawing

bancika, 2020-09-06:
Yes, I agree. Initially everything was in pixels and I started transitioning to doubles, so there could be some legacy leftovers from that era.