bancika/diy-layout-creator #553: Snap to grid for a 'net' does not function properly

TomKamphuys, 2020-06-07:
Single component do snap to the grid. However, e.g. a turret with some components attached to it, will jump the grid spacing amount when moved, but not snap to a grid point. Moving the end points of the component alone works fine.

bancika, 2020-09-02:
It depends on the selection and how components are placed. Moving more than one component at a time does not affect relative distances between each of the point in the selection. If you start with a selection where some of the points land on the grid and others don't, it's impossible to move them and snap them all to the grid because that could ruin the shape of some components. Do you have a screenshot of before and after so I can confirm that it's the case there?

TomKamphuys, 2020-09-03:
Here is the before screenshot:
[image]

Notice the center turret is not at a grid point.

Here is after a snap to grid move:
[image]

Notice that the turret has moved a grid step in x and y from the original non-grid position, but has not snapped to the grid.

bancika, 2020-09-03:
Hm and you are dragging only the turret, not the rest?

TomKamphuys, 2020-09-03:
yes, correct. As indicated by the red circle around the edge of the turret.

bancika, 2020-09-15:
done for 4.9.0

closed 2020-09-15 by bancika