bancika/diy-layout-creator #554: Shortcut for "Highlight Connected Areas"

pyscorcher, 2020-06-13:
I usually use DLC for creating pcb layouts from circuit diagrams. To check the layout I frequently use the "highlight connected areas" function to check for erros in my layout and correct them. It would be very helpful to have a shortcut for this function. My suggestion would be either:

* Have a simple shortcut to toggle between "Highlight Connected Ares" on and off, or
* Use something like [alt]+[click] for highlighting. (And normal click as usual for editing.)

Furthermore I think this is not really a "config" thing, it is more like an own tool!

bancika, 2020-09-02:
done for 4.8.0 release

closed 2020-09-02 by bancika

pyscorcher, 2020-09-06:
That is awesome, thank you very much! It improves the workflow so much, at least for me:)