bancika/diy-layout-creator #555: silly problem....or not? MAC OSX

animalejo, 2020-06-20:
First to all, i'm a very noob Mac user, i used to be a Windows user, now i'm in Mac.

My problem 01 is: 
The icons for the extension ".diy"don't show up for default, i have to put it one by one. check the video 01 ---> [LINK 01](https://drive.google.com/file/d/1S06eicZm9z7vxgVRsvHcnm7LsXLa3NN9/view?usp=sharing)
I already made the option to select the default program for that extension, but it still doesn't work

My other problem 02 is: 
That when I open a file the program starts as a new file, so I have to open that file already inside the program using the "open" option and look for it in the folder.
See video 02 --> [LINK 02](https://drive.google.com/file/d/1-whRyikiV56-yAmkdwis5JgU6kTYGx6e/view?usp=sharing)

animalejo, 2020-06-20:
forget to said, this is not happened in windows

htor, 2020-09-16:
this can also be closed, as #575 fixes this

closed 2020-09-16 by bancika