bancika/diy-layout-creator #556: Bug in board code

EL34xyz, 2020-06-30:
The new board code seems to have some sort of bug

Setting the explicit board length seems to change the length on the DIYLC screen
But then looking at the .xml file in a text editor, the length does not save

Look at my screen shot
The board is set to 11.5 inches long but the .xml file shows it as 12.97 inches long

[image]

bancika, 2020-06-30:
Yes, I think that's intentional. Length and width are saved as you can see
below and those will be honored when rendering the component. The second
control point is ignored in that case.

EL34xyz, 2020-06-30:
It's causing problems when my on line program analyzes a customers custom board.
He wants a 11.5 inch long board but the program is showing him a 12.97 inch long board

EL34xyz, 2020-06-30:
Here's I set the board length to 13 inches but it does not save the top control points

It only save the draw coords

    <diylc.boards.BlankBoard>
      <name>Board1</name>
      <alpha>127</alpha>
      <value></value>
      <controlPoints>
        <point x="0.0" y="0.0"/>
        <point x="11.5" y="3.125"/>
      </controlPoints>
      <firstPoint x="0.0" y="0.0"/>
      <secondPoint x="11.5" y="3.125"/>
      <boardColor hex="cccccc"/>
      <borderColor hex="ada47d"/>
      <coordinateColor hex="b6b6b6"/>
      <drawCoordinates>true</drawCoordinates>
      <coordinateOrigin>Top_Left</coordinateOrigin>
      <length value="13.0" unit="in"/>
      <width value="3.125" unit="in"/>
      <mode>Explicit</mode>
      <type>SQUARE</type>
    </diylc.boards.BlankBoard>

bancika, 2020-06-30:
Yes, that's because of this piece

 <mode>Explicit</mode>

the second control point has no meaning to the app. It's there only if you
go back to the other mode.

EL34xyz, 2020-06-30:
I don't understand
The board is only one length, not two
Why wouldn't both coords be set the same?

EL34xyz, 2020-06-30:
So I just tested both explicit and opposing
I set the length to 14 inches in explicit
Now I switch to opposing and it changed my board length 

I don't want some other board length after I have already set the length I want

bancika, 2020-06-30:
It depends how you look at things :)

EL34xyz, 2020-06-30:
You are kidding right?
Why even have the opposing points selection if it does not have the same coords at explicit?

bancika, 2020-06-30:
Not really. You have a board that is sized using two points. You switch it
to explicit mode and then only one point and dimensions matter. You go back
to two points, but a valid question is what the "going back" is. going back
to the original size or going back to two points while maintaining the
explicit dimensions set while the board was in that mode. One can argue
that either of those approaches is valid.

EL34xyz, 2020-06-30:
Seriously Bane?
That only makes sense in a world without logic

Who wants the program to change their board dimensions after they have already set them to a certain size using one of the methods?

EL34xyz, 2020-06-30:
So I opened an old .diy file just now that does not have the new draw coords present
The board is9 inches long

DIYLC shows it as 13 inches long
Where is that coming from?

Here's the board section in the file
```
    <diylc.boards.BlankBoard>
      <name>Hoffman Board</name>
      <alpha>127</alpha>
      <value></value>
      <controlPoints>
        <point x="1.0" y="1.3"/>
        <point x="9.0" y="4.3"/>
      </controlPoints>
      <firstPoint x="1.0" y="1.3"/>
      <secondPoint x="9.0" y="4.3"/>
      <boardColor hex="cccccc"/>
      <borderColor hex="ada47d"/>
      <coordinateColor hex="b6b6b6"/>
      <drawCoordinates>true</drawCoordinates>
      <coordinateOrigin>Top_Left</coordinateOrigin>
      <mode>TwoPoints</mode>
      <type>SQUARE</type>
    </diylc.boards.BlankBoard>
```
[image]

EL34xyz, 2020-06-30:
Here's one of my Pro Jr DIYLC files
The board was designed as is 10.06 inches long a long time ago in DIYLC
Now click back and forth between opposing points and Explicit dimensions in the board edit selection box
Which board do you think is the correct one?

[image]

[image]

bancika, 2020-06-30:
does it load correctly the first time?

EL34xyz, 2020-06-30:
That one did, but the 9 inch board up above did not
It came up as 13 inches some how

But if I drag the board out longer in order to add more components, then explicit is not correct

Also, if I switch to explicit, it wants me to enter the dimensions

It already knows the correct dimensions from the XML file
Why do I have to figure out what the dimensions are from the ruler?

The original board is 12.14 inches long x 3.125 inches high
I have to go open the xml file to figure out what the original dimensions were if I want to use explicit

Why can't the program just use the original dimensions for explicit if there never were any explicit dimensions?

bancika, 2020-06-30:
I hear what you are saying. I did my best with the time I had to do it. It
can always be improved.

EL34xyz, 2020-06-30:
Thanks
I was only made aware of this by a customer that could not get his custom board size set correctly
I am surprised no other customers mentioned this until today
But it is June and things have slowed down a bit here

bancika, 2020-09-15:
Made a change that keeps explicit dimensions in sync with control point location. Will go out with 4.9.0

closed 2020-09-15 by bancika

EL34xyz, 2020-09-15:
Awesome, thanks!

And the opposite also?
Whichever dimension you set sets the other one?

bancika, 2020-09-15:
Only if the sequence matches the selected sizing mode. If you set sizing
mode to "two points" and set the explicit length it won't do anything.