bancika/diy-layout-creator #557: can't open file

Kima378, 2020-07-03:
Can't open it. It is almost finished 😞 Program got full (red in the bottom right corner) and then completly frozen. When I reopened it now it says: 
Could not open file XXXX. Check the log for details.

Kima378, 2020-07-03:
file. change to diy
[model1_3.txt](https://github.com/bancika/diy-layout-creator/files/4868944/model1_3.txt)

bancika, 2020-07-03:
You can try getting a previous working version from your user folder, then
go do .diylc/backups

Kima378, 2020-07-03:
WOW simple but amazing! 
Thank you! I was crying on my desk for few hours :D

closed 2020-07-03 by Kima378

htor, 2020-08-16:
> You can try getting a previous working version from your user folder, then go do .diylc/backups

did not know it kept backups! could we place this info in the main README? i see you maintain a wiki branch, but wouldn't it be better to have those files sitting in the master branch? a bit confusing when digging through the repo for more info. :-)