bancika/diy-layout-creator #561: DIYLC 4.7.0 Will Not Open in Windows

BrotherBones, 2020-07-12:
Hi Bane, 

I'm having an issue with my DIYLC on start up.   I had been running 4.1.0 through 7/11/2020 when it stopped opening suddenly.  I've attached log for the last instance it worked and then when it wouldn't open.  I had made a schematic of a project and saved it.  When I went to open the PCB layout file I had it was stuck on the "splash" png.  Only way to make the splash screen go away is to "End" the Java application in the task manager.

Last working instance log:
[log-2020-07-11 07.18.38.txt](https://github.com/bancika/diy-layout-creator/files/4909413/log-2020-07-11.07.18.38.txt)

First instance not-working log: 
[log-2020-07-11 11.02.33.txt](https://github.com/bancika/diy-layout-creator/files/4909415/log-2020-07-11.11.02.33.txt)

I had not done any updates to my computer at that time so not sure why it suddenly decided to not load. Today when reviewing I discovered I needed to update my Windows (major update), Java, and DIYLC to 4.7.0.  I updated all to the most current versions available and it still presents this issue.  I also tried downloading 4.1.0 (since it worked before) and 4.6.0 both installing directly to the computer as well as running locally from the file folder exe after extracting the zip file.  They both exhibit the same issue of being stuck on the splash screen.  

Also went through and did registry fix and clean and tried installing an older version of JAVA 7 to see if that would work.  Still nothing. Running as administrator didn't work either.  On a whim I tried my previous DIYLC 3.31.0 that was still in my files.  That one will open, but nothing I've created since switching to 4.1.0 will open in this version.   

I then started from scratch and deleted DIYLC and JAVA and reinstalled most current versions (4.7.0 and 8u251).  It's still stuck on splash screen so here submitting this issue.  Here is the log of trying to open the current install and pic of my windows specs.  Let me know if you need more info.  

[log-2020-07-12 04.51.28.txt](https://github.com/bancika/diy-layout-creator/files/4909500/log-2020-07-12.04.51.28.txt)

Much props for this program by the way.  It's been a great tool for myself and pedal making.  This is the first issue I've ever run into with it.   

Cheers, 
Blake

[image]

bancika, 2020-07-12:
Can you try to rename config.xml file to something else? You can find it in
c:\users\<your name>\.diylc

That will wipe your old DIYLC configuration and start over.

Cheers

BrotherBones, 2020-07-12:
Alright!  That did the trick.  Thanks for the answer! Cheers and stay safe!  -B

bancika, 2020-07-12:
Nice!

closed 2020-09-15 by bancika