bancika/diy-layout-creator #563: Hook up Wires

BiffDebris, 2020-08-08:
My Windows Home Edition and Java are current. I can't get the hook-up wires to move to where I need them to go. I've used this program before without problems. Is this something known?
Thank you for reporting issues and contributing ideas and features to DIYLC repository!

Before you proceed, please read the following:

* If you are reporting a bug, please attach the latest application log file together with description of steps that lead to the issue. They can be found in the user directory under `.diylc/logs` directory. On Windows 7 and later, it's located in `c:/Users/<your username>/.diylc/logs`. On Linux and MacOS, it's typically in `/Users/<your username>/.diylc/logs`.
* If you are having issues starting the application under MacOS, make sure that 3rd party apps are allowed on your Mac. This document covers how to do it - https://support.apple.com/en-us/HT202491.
* Try to look for similar unresolved issues that are already reported to avoid duplicates.
* Make sure that you are running the latest version of DIYLC.

EL34xyz, 2020-08-10:
No problems here on Windows 10
I grab the green points and shape the wires just fine

[image]

bancika, 2020-08-10:
Must be something related to that particular project. Can you upload the
project file (just rename it from .diy to .txt)? Does it happen if you
start a new project or open some other one?

dcthorson, 2020-12-31:
@BiffDebris : Did you confirm your snap settings (`Config` --> `Snap To`)?