bancika/diy-layout-creator #564: App Stability Issue in OS X Catalina

Fith76, 2020-08-12:
Every time I run DIYLC and open a png or jpg or any other type of image file in Pixelmator, or launch any other application that uses Java, DIYLC will stop responding.  Most of the time, it will also kill the keyboard and trackpad.  I am attaching the error log presented after a forced reboot.  It has to do with  the JavaAppLauncher.

Any help would be greatly appreciated!  I use this software on a daily basis.

[DIYLC-JAVA Crash.txt](https://github.com/bancika/diy-layout-creator/files/5060036/DIYLC-JAVA.Crash.txt)

htor, 2020-09-08:
it's most likely that the Searcable Tree is the cause for unresponsiveness on mac. try turning it off (Config > Toolbox > Tabbed Toolbar) and see if the problem persists.

bancika, 2020-09-16:
Does it happen if you switch the toolbox from the side tree to the tabbed bar (from the config)?

Fith76, 2020-09-17:
The problem does not persist if I switch the toolbox to tabbed bar.

Jeremy Le-Roy
jeremy@leroy.com

> 
> 
> Does it happen if you switch the toolbox from the side tree to the tabbed bar (from the config)?

Jeremy Le-Roy
jeremy@leroy.com

bancika, 2020-09-17:
Thanks, that helps.

htor, 2020-09-17:
i did some rudimentary debugging and it looks like the calls to `process()` take a very long time (growing exponentially for each letter typed in the search field) 

tried commenting out parts of process() and the interface was responsive at least (with a non-functional search tree): https://github.com/bancika/diy-layout-creator/blob/07ba8f14f7503825e32d04b2a4a85bbd4a9f4873/diylc/diylc-swing/src/org/diylc/swing/plugins/tree/TreePanel.java#L680

bancika, 2020-09-17:
Can you give this version a shot and see if it's any better on mac? Thanks!

[diylc-4.9.0-osx.zip](https://github.com/bancika/diy-layout-creator/files/5240179/diylc-4.9.0-osx.zip)
[diylc-4.9.0.zip](https://github.com/bancika/diy-layout-creator/files/5240180/diylc-4.9.0.zip)

htor, 2020-09-17:
i tested that build and wow! it does really fix the lagging problem. no lagging whatsoever when enabled Searchable Tree.
tried opening a file and dragging things from the tree. no problem or application lagging.

:-)

bancika, 2020-09-17:
Nice! Yes, I rebuilt the model for that tree from ground up instead of using the canned solution. It should be as fast as it can be. Just wondering if that was the root cause of issues on macs.

bancika, 2020-09-17:
It would be great @Fith76 could verify that this solves his problem too.

closed 2020-11-19 by bancika