bancika/diy-layout-creator #565: Cursor won't work with MacOSX Catalina 10.15.5

secretriches, 2020-08-13:
I just downloaded and installed diy-layout-creator and current version of Java on my iMac. I can open the program and create a new board but then I can't get the cursor to do anything with the board or other components. It's totally unresponsive. The board can be highlighted and edited (using the right click on the mouse) but, there is no ability to do anything beyond this point. Is this a possible bug in the program?

htor, 2020-08-16:
maybe check in ~/.diylc/logs/ for the newest log file and see if there are any errors? what is your java version?

bancika, 2020-08-16:
unlikely a bug if such a simple case doesn't work. Many use the app.

can you specify *exactly* what you are doing? Also, try resetting the
config by renamin config.xml file in .diylc directory in your user config,
in case you switched some settings unknowingly.

htor, 2020-09-08:
it's most likely that the Searcable Tree is the cause for unresponsiveness on mac. try turning it off (Config > Toolbox > Tabbed Toolbar) and see if the problem persists.

bancika, 2020-12-11:
good chance this was fixed. Please comment if still an issue.

closed 2020-12-11 by bancika