bancika/diy-layout-creator #566: macos: nothing happens when opening .diy files

htor, 2020-08-16:
files with .diy extension does not associate properly with DIYLC.app and when opening them with double click or using "open with ..." or dragging the file into DIYLC then nothing happens. only if i open DIYLC first and use its own open dialog it will open files.

looks like:
[image]

i have got macos to remember a default application for that file extension (select file > cmd+alt+i > open with ... > change all...) but it does not help. 

[image]

i've checked the log file for errors, but there is nothing there.

bancika, 2020-08-17:
seems to be another peculiarity of macos. On all other OS when you
associate a file with an app, file name is passed as the first argument to
the application. Seems that it's not the case here, since it doesn't work.

htor, 2020-08-17:
yeah, getting the app packaged correctly can be tricky when making cross-platform apps.

htor, 2020-09-03:
and i just want to say thanks for making this fine software. many diy projects wouldn't be possible without it!

htor, 2020-09-06:
okay, so i managed to build the project on my mac, tweak the `build.xml` so that the DIYLC.app bundle registers itself correctly with regards to the custom `.diy` file type (and it is now in the list of recommended applications in the drop down). it looks like this:

[image]

the remaining problem is that the `main()` method does not receive any arguments upon double-clicking .diy files. after googling i found out that on mac, your app bundle does not indeed receive file arguments. it seems you need to listen for an "open file" event and register a handler for that.
https://stackoverflow.com/questions/10546427/java-swing-how-to-double-click-a-project-file-on-mac-to-open-my-application-an

maybe that last SO answer could work?
can we use java9 apis? or is java8 the limit?
i've had only success building the project with oracle jdk8.

bancika, 2020-09-07:
I merged and checked it out and as you said, it doesn't even compile on windows or linux. That's a big problem, we cannot have two versions of the code for Mac and non-Mac, and in that case I wouldn't be able to compile the mac version :) 

Maybe we can reference this dummy apple utility lib that has empty implementations of all those apple-specific classes. I tried patching your code on next branch and now it compiles and runs on windows. Can you try it on a mac?

[image]

htor, 2020-09-07:
i see!

https://github.com/bancika/diy-layout-creator/commit/51645012289574f937ecf67c4eed235854d6d601 builds and runs fine on my mac

👍

bancika, 2020-09-07:
> 5164501

Cool, we just need to make sure that the dummy library doesn't end up in
the build and get used instead of the real thing on mac. Did you try
running the built version and opening DIY files?

Thanks

htor, 2020-09-07:
i rebuilt all three packages on your next branch and tested with opening DIY files. works exactly the same as my PR.

bancika, 2020-09-07:
awesome, thanks! I'll add it to 4.9.0 release

PR #575 (2020-09-07): fix file extension and file opening for macos
this PR fixes #566 and #370 

i managed to build the project on my mac, tweak the `build.xml` so that the DIYLC.app bundle registers itself correctly with regards to the custom `.diy` file type (and it is now in the list of recommended applications in the drop down with correct icon). it looks like this:

[image]

the remaining problem is that the `main()` method does not receive any arguments upon double-clicking .diy files. after googling i found out that on mac, your app bundle does not indeed receive file arguments. it seems you need to listen  [... cut]
closed 2020-09-07 by bancika