bancika/diy-layout-creator #567: feature: zoom hotkeys, zoom slider

htor, 2020-08-16:
i find it unpractical to always use the dropdown menu to select zoom level. i could have used my mouse, but currently touchpad zooming is currently broken on macos (#501).

bancika, 2020-08-16:
ctrl + mouse scroll should work on all OSs. Maybe not macs, they are always
special and quirky.

bancika, 2020-09-16:
Added keyboard shortcuts for 4.9.0

htor, 2020-09-16:
hey, this is great!

htor, 2020-09-18:
i tested the 4.9.0 release. weirdly, zooming still does not work with hotkeys.
tried CMD-+ and CTRL-+ but none do anything to zoom. hmm

bancika, 2020-09-18:
Maybe it's mac again :) You are in the canvas, right? No other UI component is selected?

htor, 2020-09-18:
yeah, 
i'm in a blank canvas without anything selected.
i don't know what's wrong, but _only_ CTRL+0 works, after zooming using the drop-down menu. 

also, i was hoping for cmd + cmd -, cmd+0 on mac instead of CTRL hotkeys. the CMD key (somtimes called "command" or "meta").

bancika, 2020-09-18:
I can add meta key

htor, 2020-11-20:
hi, just tested the latest release, and CMD+ and CMD - does not zoom in or out. CMD 0 does work and resets the zoom.

bancika, 2020-12-09:
hmm. Are you using +/- keys that are dedicated keys or shared with - and =?
I suspect it's the latter, in which case, those are totally different keys
:)

htor, 2020-12-09:
> hmm. Are you using +/- keys that are dedicated keys or shared with - and =? I suspect it's the latter, in which case, those are totally different keys :)
> […](#)
> -- Branislav Stojkovic [www.diy-fever.com](http://www.diy-fever.com)

i'm using the one shared with + and =. i guess you're referring to the numpad in the former case? macbooks don't have numpads you know! :-)

htor, 2020-12-09:
i'll gladly test a build whenever

bancika, 2020-12-09:
Here it is, together with pinch zoom

https://drive.google.com/file/d/1Rm2JzBqa64oikgU0g66ySFy_v1G_CCsh/view?usp=sharing

htor, 2020-12-09:
wow, nice! 

tested, and 0 and - works perfect. + is triggered on wrong keys. i left a comment on the commit! maybe you could try the suggestion?

otherwise, pinch also works on mac! great :-)

bancika, 2020-12-10:
let's try with this
https://drive.google.com/file/d/1Rm2JzBqa64oikgU0g66ySFy_v1G_CCsh/view?usp=sharing

htor, 2020-12-10:
that did the trick.
now zooming in out and reset works perfectly!

bancika, 2020-12-10:
Excellent!

closed 2020-12-11 by bancika