bancika/diy-layout-creator #568: text character width too tight in PDF export

htor, 2020-08-16:
when exporting a layout to PDF the spacing between characters in text appears is too tight - resulting in bad readability, especially when zoomed out.

normal in DIYLC:

[image]

[image]

exported PDF:

[image]

[image]

macos 10.13
diylc 4.7.0

bancika, 2020-09-14:
does it happen with any font or only certain ones?

htor, 2020-09-16:
some fonts it seems have better spacing and some will have too narrow spacing. 

i noticed some group of fonts all result in the same PDF rendering regardless of selected font under edit > project settings. looks like it defaults to some standard PDF font during export in those cases?

for instance selecting `.SF NS Text`, `Menlo` or `Courier New` will convert my canvas text into that font, but exporting will use another "sans serif" looking font, with the above narrow spacing. it looks like the `Dialog` font is used somtimes by default on drawn text boxes also.

bancika, 2020-09-16:
Can you please try the latest preview version and see if it helps? I switched to another vendor for PDF conversion. It's lightweight and allows me to render fonts to vectors, to avoid any interpretation issues and make sure that it's 100% as rendered in the app.

[diylc-4.9.0.zip](https://github.com/bancika/diy-layout-creator/files/5233164/diylc-4.9.0.zip)

[diylc-4.9.0-osx.zip](https://github.com/bancika/diy-layout-creator/files/5233162/diylc-4.9.0-osx.zip)

htor, 2020-09-16:
ah, can't get it to start, it hangs at splash screen:

```txt
18:30:52,539  INFO [main] ConfigurationManager - Initializing configuration
18:30:52,775  INFO [main] ConfigurationManager - Saving configuration
18:30:52,862  INFO [main] DIYLCStarter - Loading languages...
18:30:52,868  INFO [main] LangUtil - Loading translation for English
18:30:52,887 ERROR [main] LangUtil - Error loading language file
java.nio.file.NoSuchFileException: lang/English.txt
	at sun.nio.fs.UnixException.translateToIOException(UnixException.java:86)
	at sun.nio.fs.UnixException.rethrowAsIOException(UnixException.java:102)
	at sun.nio.fs.UnixException.rethrowAsIOException(UnixException.java:107)
	at sun.nio.fs.UnixFileSystemProvider.newByteChannel(UnixFileSystemProvider.java:214)
	at java.nio.file.Files.newByteChannel(Files.java:361)
	at java.nio.file.Files.newByteChannel(Files.java:407)
	at java.nio.file.spi.FileSystemProvider.newInputStream(FileSystemProvider.java:384)
	at java.nio.file.Files.newInputStream(Files.java:152)
	at java.nio.file.Files.newBufferedReader(Files.java:2784)
	at java.nio.file.Files.lines(Files.java:3744)
	at org.diylc.lang.LangUtil.load(Unknown Source)
	at org.diylc.lang.LangUtil.configure(Unknown Source)
	at org.diylc.DIYLCStarter.main(Unknown Source)
18:30:52,889 DEBUG [main] DIYLCStarter - Java version: 1.8.0_261-b12 by Oracle Corporation
18:30:52,889 DEBUG [main] DIYLCStarter - OS: Mac OS X 10.13.6
18:30:52,889  INFO [main] DIYLCStarter - Starting DIYLC with working directory /Users/htor
18:30:52,890  INFO [main] DIYLCStarter - Configuration dump start.
18:30:52,890  INFO [main] DIYLCStarter - antiAliasing = true
18:30:52,890  INFO [main] DIYLCStarter - hiQualityRender = false
18:30:52,890  INFO [main] DIYLCStarter - exportGrid = false
18:30:52,891  INFO [main] DIYLCStarter - stickyPoints = false
18:30:52,891  INFO [main] DIYLCStarter - metric = true
18:30:52,891  INFO [main] DIYLCStarter - snapTo = Grid
18:30:52,891  INFO [main] DIYLCStarter - continuousCreation = false
18:30:52,891  INFO [main] DIYLCStarter - wheelZoom = false
18:30:52,891  INFO [main] DIYLCStarter - outline = false
18:30:52,892  INFO [main] DIYLCStarter - showRulers = false
18:30:52,892  INFO [main] DIYLCStarter - showGrid = true
18:30:52,892  INFO [main] DIYLCStarter - highlightContinuityArea = false
18:30:52,892  INFO [main] DIYLCStarter - hardwareAcceleration = true
18:30:52,892  INFO [main] DIYLCStarter - extraSpace = true
18:30:52,892  INFO [main] DIYLCStarter - renumberOnPaste = false
18:30:52,892  INFO [main] DIYLCStarter - defaultTemplatesImported = true
18:30:52,892  INFO [main] DIYLCStarter - defaultBlocksImported = true
18:30:52,892  INFO [main] DIYLCStarter - Configuration dump end.
18:30:52,895 DEBUG [Thread-1] DIYLCStarter - Starting font pre-loading
18:30:52,902 DEBUG [Thread-1] ResourceLoader - Detected OSX bundle path: /Users/htor/Downloads/DIYLC2.app
18:30:52,902 DEBUG [Thread-1] ResourceLoader - Looking for files in: /Users/htor/Downloads/DIYLC2.app/Contents/Resources/fonts
18:30:52,903  INFO [Thread-1] DIYLCStarter - Dynamically loading font: square721_bt.ttf
18:30:53,086  INFO [main] Presenter - Current DIYLC version: 4.9.0
18:30:54,091  INFO [Thread-1] DIYLCStarter - Dynamically loading font: consola.ttf
18:30:54,092  INFO [Thread-1] DIYLCStarter - Dynamically loading font: lucon.ttf
18:30:54,343 DEBUG [Thread-1] DIYLCStarter - Finished font pre-loading
```

bancika, 2020-09-16:
Let's try one more time.

[diylc-4.9.0-osx.zip](https://github.com/bancika/diy-layout-creator/files/5234551/diylc-4.9.0-osx.zip)
[diylc-4.9.0.zip](https://github.com/bancika/diy-layout-creator/files/5234552/diylc-4.9.0.zip)

htor, 2020-09-16:
nice!
the rendering looks very good now in this build:

[image]
[image]

the problem is gone. tried with a couple of fonts, but no issue with spacing. looks pixel identical!

bancika, 2020-09-16:
Cool. Thanks for checking it out. I'll include with the next version.

htor, 2020-09-17:
thanks for looking into it

closed 2020-09-17 by bancika