bancika/diy-layout-creator #569: Unable to move large number of components

robrobinette, 2020-08-25:
When I highlight a large number of schematic components, such as 2/3 of my JCM800 6V6 schematic, I cannot move the components. The components are highlighted but the arrow keys do nothing and if I try to drag with the mouse the components are un-highlighted and only the component directly under the mouse moves. Smaller groups of components in the same schematic can be moved with the arrows and mouse. I have had this issue for quite some time and with other complex DIY files.

I am using the latest version of DIYLC and Win 10 x64

The JCM800 6V6 schematic: [https://robrobinette.com/images/Guitar/Marshall/RR2104_Master_Volume_Micro/RR2104_Master_Volume_6V6_Schematic.diy](url)

14:22:25,925  INFO [main] ConfigurationManager - Initializing configuration
14:22:26,247  INFO [main] LangUtil - Loading translation for English
14:22:26,271 DEBUG [main] DIYLCStarter - Java version: 1.8.0_201-b09 by Oracle Corporation
14:22:26,271 DEBUG [main] DIYLCStarter - OS: Windows 10 10.0
14:22:26,271  INFO [main] DIYLCStarter - Starting DIYLC with working directory C:\Program Files (x86)\DIYLC
14:22:26,281 DEBUG [Thread-3] DIYLCStarter - Starting font pre-loading
14:22:26,293 DEBUG [Thread-3] ResourceLoader - Looking for files in: fonts
14:22:26,293  INFO [Thread-3] DIYLCStarter - Dynamically loading font: consola.ttf
14:22:26,737  INFO [main] Presenter - Current DIYLC version: 4.7.0
14:22:26,739  INFO [Thread-3] DIYLCStarter - Dynamically loading font: lucon.ttf
14:22:26,741  INFO [Thread-3] DIYLCStarter - Dynamically loading font: square721_bt.ttf
14:22:26,750  INFO [main] Presenter - Checking if variants need to be updated
14:22:26,750  INFO [main] Presenter - Loading component types.
14:22:26,751 DEBUG [main] Utils - Scanning jar: /C:/Program Files (x86)/DIYLC/lib/diylc-core.jar
14:22:26,774 DEBUG [main] Utils - Scanning jar: /C:/Program Files (x86)/DIYLC/library/main.jar
14:22:26,909 DEBUG [main] JarScanner - Found JAR file: main.jar
14:22:26,909 DEBUG [main] JarScanner - Scanning main.jar
14:22:26,948  INFO [main] ClassLoaderUtil - Adding file to the classpath: C:\Program Files (x86)\DIYLC\library\main.jar
14:22:26,950 DEBUG [main] JarScanner - Loaded class: org.diylc.components.Abstract3LegSymbol
14:22:26,951 DEBUG [main] JarScanner - Loaded class: org.diylc.components.AbstractLeadedComponent
14:22:26,951 DEBUG [main] JarScanner - Loaded class: org.diylc.components.AbstractMultiPartComponent
14:22:26,951 DEBUG [main] JarScanner - Loaded class: org.diylc.components.AbstractRadialComponent
14:22:26,951 DEBUG [main] JarScanner - Loaded class: org.diylc.components.AbstractSchematicLeadedSymbol
14:22:26,951 DEBUG [main] JarScanner - Loaded class: org.diylc.components.boards.AbstractBoard
14:22:26,951 DEBUG [main] JarScanner - Loaded class: org.diylc.components.boards.BlankBoard
14:22:26,952 DEBUG [main] JarScanner - Loaded class: org.diylc.components.boards.Breadboard
14:22:26,952 DEBUG [main] JarScanner - Loaded class: org.diylc.components.boards.EyeletBoard
14:22:26,952 DEBUG [main] JarScanner - Loaded class: org.diylc.components.boards.MarshallPerfBoard
14:22:26,952 DEBUG [main] JarScanner - Loaded class: org.diylc.components.boards.PerfBoard
14:22:26,952 DEBUG [main] JarScanner - Loaded class: org.diylc.components.boards.ProtoBoard
14:22:26,952 DEBUG [main] JarScanner - Loaded class: org.diylc.components.boards.TagStrip
14:22:26,952 DEBUG [main] JarScanner - Loaded class: org.diylc.components.boards.TerminalStrip
14:22:26,953 DEBUG [main] JarScanner - Loaded class: org.diylc.components.boards.TriPadBoard
14:22:26,953 DEBUG [main] JarScanner - Loaded class: org.diylc.components.boards.VeroBoard
14:22:26,953 DEBUG [main] JarScanner - Loaded class: org.diylc.components.chassis.ChassisPanel
14:22:26,953 DEBUG [main] JarScanner - Loaded class: org.diylc.components.chassis.EllipticalCutout
14:22:26,953 DEBUG [main] JarScanner - Loaded class: org.diylc.components.chassis.PolygonalCutout
14:22:26,953 DEBUG [main] JarScanner - Loaded class: org.diylc.components.chassis.RectangularCutout
14:22:26,953 DEBUG [main] JarScanner - Loaded class: org.diylc.components.connectivity.CopperTrace
14:22:26,953 DEBUG [main] JarScanner - Loaded class: org.diylc.components.connectivity.CurvedTrace
14:22:26,954 DEBUG [main] JarScanner - Loaded class: org.diylc.components.connectivity.CutLine
14:22:26,954 DEBUG [main] JarScanner - Loaded class: org.diylc.components.connectivity.Dot
14:22:26,954 DEBUG [main] JarScanner - Loaded class: org.diylc.components.connectivity.Eyelet
14:22:26,954 DEBUG [main] JarScanner - Loaded class: org.diylc.components.connectivity.GroundFill
14:22:26,954 DEBUG [main] JarScanner - Loaded class: org.diylc.components.connectivity.HookupWire
14:22:26,954 DEBUG [main] JarScanner - Loaded class: org.diylc.components.connectivity.Jumper
14:22:26,954 DEBUG [main] JarScanner - Loaded class: org.diylc.components.connectivity.Line
14:22:26,955 DEBUG [main] JarScanner - Loaded class: org.diylc.components.connectivity.SolderPad
14:22:26,955 DEBUG [main] JarScanner - Loaded class: org.diylc.components.connectivity.TraceCut
14:22:26,955 DEBUG [main] JarScanner - Loaded class: org.diylc.components.connectivity.Turret
14:22:26,955 DEBUG [main] JarScanner - Loaded class: org.diylc.components.connectivity.TwistedWire
14:22:26,955 DEBUG [main] JarScanner - Loaded class: org.diylc.components.electromechanical.BatterySnap9V
14:22:26,955 DEBUG [main] JarScanner - Loaded class: org.diylc.components.electromechanical.CliffJack1_4
14:22:26,955 DEBUG [main] JarScanner - Loaded class: org.diylc.components.electromechanical.ClosedJack1_4
14:22:26,955 DEBUG [main] JarScanner - Loaded class: org.diylc.components.electromechanical.DIPSwitch
14:22:26,956 DEBUG [main] JarScanner - Loaded class: org.diylc.components.electromechanical.FuseHolderPanel
14:22:26,956 DEBUG [main] JarScanner - Loaded class: org.diylc.components.electromechanical.IECSocket
14:22:26,956 DEBUG [main] JarScanner - Loaded class: org.diylc.components.electromechanical.MiniRelay
14:22:26,956 DEBUG [main] JarScanner - Loaded class: org.diylc.components.electromechanical.MiniToggleSwitch
14:22:26,956 DEBUG [main] JarScanner - Loaded class: org.diylc.components.electromechanical.OpenJack1_4
14:22:26,956 DEBUG [main] JarScanner - Loaded class: org.diylc.components.electromechanical.PilotLampHolder
14:22:26,956 DEBUG [main] JarScanner - Loaded class: org.diylc.components.electromechanical.PinHeader
14:22:26,957 DEBUG [main] JarScanner - Loaded class: org.diylc.components.electromechanical.PlasticDCJack
14:22:26,957 DEBUG [main] JarScanner - Loaded class: org.diylc.components.electromechanical.RCAJack
[... 360 lines cut ...]
14:24:33,053 DEBUG [AWT-EventQueue-0] Presenter - mouseClicked(java.awt.Point[x=288,y=1019], 1, false, false, false)
14:24:36,150  INFO [AWT-EventQueue-0] ActionFactory - SaveAction triggered
14:24:36,151 DEBUG [SwingWorker-pool-1-thread-4] ActionFactory - Saving to C:\Users\Rob\Documents\Web Page Publish\robrobinette\images\Guitar\AA1164\Princeton_Reverb_Micro\Princeton Reverb Micro_Layout.diy
14:24:36,151  INFO [SwingWorker-pool-1-thread-4] Presenter - saveProjectToFile(C:\Users\Rob\Documents\Web Page Publish\robrobinette\images\Guitar\AA1164\Princeton_Reverb_Micro\Princeton Reverb Micro_Layout.diy)
14:24:36,152  INFO [SwingWorker-pool-1-thread-4] ProjectFileManager - saveProjectToFile(C:\Users\Rob\Documents\Web Page Publish\robrobinette\images\Guitar\AA1164\Princeton_Reverb_Micro\Princeton Reverb Micro_Layout.diy)
14:24:36,178  INFO [SwingWorker-pool-1-thread-4] ConfigurationManager - Saving configuration
14:24:44,886  INFO [AWT-EventQueue-0] ActionFactory - ExportPDFAction triggered
14:24:45,951 DEBUG [AWT-EventQueue-0] DrawingManager - Render stats: {connectivity.HookupWire=7068x 733ms; passive.Resistor=2451x 176ms; misc.Label=11913x 155ms; connectivity.Eyelet=4104x 124ms; passive.RadialCeramicDiskCapacitor=342x 90ms; passive.AxialFilmCapacitor=513x 81ms; connectivity.Jumper=513x 66ms; passive.AxialElectrolyticCapacitor=399x 24ms; tube.TubeSocket=342x 23ms; passive.PotentiometerPanel=342x 21ms; connectivity.SolderPad=1083x 19ms; electromechanical.MiniToggleSwitch=171x 7ms; shapes.Ellipse=171x 5ms; connectivity.Dot=399x 5ms; misc.GroundSymbol=114x 3ms; passive.MultiSectionCapacitor=57x 3ms; shapes.Rectangle=57x 2ms; boards.BlankBoard=57x 1ms}
14:24:45,951 DEBUG [AWT-EventQueue-0] DrawingManager - Page stats: 57x 159ms avg
[... cut ...] dows 7 and later, it's located in `c:/Users/<your username>/.diylc/logs`. On Linux and MacOS, it's typically in `/Users/<your username>/.diylc/logs`.
* If you are having issues starting the application under MacOS, make sure that 3rd party apps are allowed on your Mac. This document covers how to do it - https://support.apple.com/en-us/HT202491.
* Try to look for similar unresolved issues that are already reported to avoid duplicates.
* Make sure that you are running the latest version of DIYLC.

bancika, 2020-08-26:
Will check it, Rob.

bancika, 2020-09-02:
Seems that this component is a culprit. When attempting to move components, the system tries to figure out if the action will cause any control points to collide and cause the component to reduce to singularity. In those cases it does not allow the action to proceed. However in this case there's a line that is already reduced to singularity somehow and it prevents the selection to be moved.

I'll think if there's a way to relax this restriction. In the meantime, deleting this component from the file directly should help.

<diylc.connectivity.Line>
      <name>LN74</name>
      <alpha>127</alpha>
      <points>
        **<point x="9.0" y="0.93"/>
        <point x="9.0" y="0.93"/>**
        <point x="9.0" y="0.98"/>
      </points>

pedalmeat, 2020-12-06:
I have also experienced this. It happens when a trace, etc. has the same start- and end-points. I usually try to find it by selecting half of the layout and try to move it. Repeat with selecting and moving half of previous selection until you find it.

bancika, 2020-12-06:
Yes, that's the root cause. The algorithm is trying to prevent that
scenario and that's why you are unable to move the components. It detects
that a zero-length component will be there if the action is allowed and it
blocks it. Doesn't check if it's already zero-length.

So the main question is - how did that component become zero-length to
begin with. That happened somehow before attempting to move the large
selection and it's not supposed to happen :)

pedalmeat, 2020-12-07:
It usually happens when i select a number of components with traces attached to their ends and drag them in the direction of a trace so it gets reduced to zero length without me noticing it. It's then hidden underneath another trace or a solderpad.

bancika, 2020-12-11:
Fixed for 4.12.0

closed 2020-12-11 by bancika