bancika/diy-layout-creator #570: wishlist 6,3mm jack symbol

KENTUCKYTUKKER, 2020-08-26:
could you implement a symbol for a 6,3mm mono jack in the schematic library?

bancika, 2020-08-26:
it's already there.

KENTUCKYTUKKER, 2020-08-26:
> <diy-layout-creator@noreply.github.com>
> <author@noreply.github.com>
> (#570)

Hello,

I did find it as a physical symbol but not as a schematic symbol. Where 
can I find it then?

Thanks,

René Mikkers
------ Origineel bericht ------

KENTUCKYTUKKER, 2020-08-27:
Hello Bancika,

I did mean this symbol. See attachment. Ofcourse the resistor will not 
be part of it.

Best regards,

rené Mikkers

------ Origineel bericht ------

KENTUCKYTUKKER, 2020-09-02:
Hello Bancika,

Please inform me where I can find the Electrical Symbol in the DIY 
library. I can only find the physical symbols.

Thanks,

René

------ Origineel bericht ------
Van: "Rene Mikkers" <r.mikkers@home.nl>
Aan: "bancika/diy-layout-creator" 
<reply@reply.github.com>
Verzonden: 27-8-2020 10:21:02
Onderwerp: Re[2]: [bancika/diy-layout-creator] wishlist 6,3mm jack 
symbol (#570)

bancika, 2020-09-02:
There is none, I read it wrong the first time.

KENTUCKYTUKKER, 2020-09-02:
Hello Bane,

Thanks for the response. Is there any chance you will add this symbol in 
the near future? I did also mis the electrical symbol for a pilot light 
(circle with cross) I made one by myself but would be nice if this was 
available in the library.

kindest regards,

René

------ Origineel bericht ------

Verzonden: 2-9-2020 11:04:43
(#570)

bancika, 2020-09-17:
added jack for 4.9.0

closed 2020-09-17 by bancika

KENTUCKYTUKKER, 2020-09-18:
Hello Bane,

Thank you very much or adding the symbol or the jack.

Best regards,

René

------ Origineel bericht ------

Verzonden: 17-9-2020 12:18:38
(#570)

KENTUCKYTUKKER, 2020-09-18:
Hello Bane,

I found that you added the 6,3mm jack symbol. I noted that this is a non 
switchable socket. It would be nice to have this symbol also available 
as I send you by my mail dated 27-08-2020(see attachment) as this is 
often used within guitar amp schematics

Kindest regards, René

------ Origineel bericht ------
Van: "Rene Mikkers" <r.mikkers@home.nl>
Aan: "bancika/diy-layout-creator" 
<reply@reply.github.com>
Verzonden: 18-9-2020 09:02:35
Onderwerp: Re[2]: [bancika/diy-layout-creator] wishlist 6,3mm jack 
symbol (#570)

bancika, 2020-09-18:
That's one of the reasons I didn't do it before. It's an endless rabbit
hole with a bunch of variations.

KENTUCKYTUKKER, 2020-09-19:
Hi Bane,

Please note that there are only 2 versions. Either switchable or 
non-switchable. The stereo versions will not be used in guitar 
schematics anyway. I you would just have the switchable version, the 
user can decide by himself how to hook it up. So that version would be 
more versatile.

Kindest regards,

René

------ Origineel bericht ------

Verzonden: 18-9-2020 20:35:30
(#570)

bancika, 2020-09-19:
Not only two. You can have switches on any or all of the 3 terminals on a
stereo or 2 terminals on a mono jack. And switches can be normally open or
normally closed. There's probably dozen combinations there and it already
took me half of the day to make these two.

KENTUCKYTUKKER, 2020-09-19:
Besides the stereo versions I have never seen mono jacks with NO 
contacts. The only mono jack I have seen have NC contacts which will 
open when a jack is installed. All guitar amp schematics I came up to 
have switched sockets at the input circuitry and sometime at the speaker 
load connections. I have completed several schematics with the aid of 
your program but I'm unable to fish these concerning the input 
connections. Until now I added some notes to these schematics with the 
info Tip or Sleeve.

Kindest regards,

René

------ Origineel bericht ------

Verzonden: 19-9-2020 16:15:57
(#570)