bancika/diy-layout-creator #571: DIY LC crashing in Windows 7

absrec, 2020-08-29:
I’ve had this happen to me at least a dozen times since updating to the current version. Everything just freezes and if I press the hot key to save, the next time I open the project, it give me an error saying that it is unable to open/read the file. 

The first time was after pressing undo (CTRL-Z) a bunch of times in a row. Since then, it has happened so many times that I have begun to save my project under several different names periodically to assure that, at very least, I don’t have to start over from the beginning.

I do like the way the new version keeps traces underneath the components so there is less moving back required. Newly added components are nice too. The crashing issue makes me want to go back to an earlier version despite having to deal with some of the issues that are fixed in the new version.

Thanks for listening. :)

bancika, 2020-09-02:
Can you include the log file from the time the crash happened? That's the only way for me to know what happened.
I added a layer of protection when writing to file in the upcoming version, but still would like to know. It shouldn't happen that often.

bancika, 2021-07-11:
This was fixed in 4.16

closed 2021-07-11 by bancika