bancika/diy-layout-creator #572: Donation text at the top of the screen takes up too much vertical space when DIYLC window is narrow

Buran1997, 2020-09-04:
I'm running v4.7.0. I'm working on a laptop, and I like to have DIYLC on one half of the screen and a schematic on the other half. There's text at the top of the screen reading "Enjoying DIYLC? Click here to buy me a coffee :)". When I resize the window horizontally, this text squashes and displays vertically, like this:

Enjoying
DIYLC?
Click
here to
buy me
a
coffee
:)

[This is what it looks like](https://imgur.com/a/OugczQA) on my screen.

This causes the toolbar to expand vertically, taking up a significant portion of the window. Maybe this text could be replaced with a donate button, or it could be forced to render as a single line no matter the width of the window?

bancika, 2020-09-04:
Hm, I've never seen it like that before :) You use the app with the window
that small?

Buran1997, 2020-09-04:
> Hm, I've never seen it like that before :) You use the app with the window that small?
I do! I like having the schematic in front of me while I work. I'm on a laptop so the screen is pretty small already, but I find that it works okay for me.

Speaking of being on a laptop, a lot of web pages allow me to scroll sideways using the touchpad, but DIYLC doesn't. Is it possible to implement this?

bancika, 2020-09-10:
fixed the donation link, please open a separate issue for scrolling

closed 2020-09-10 by bancika