bancika/diy-layout-creator #574: Grid spacing issue, possibly related to #552

pmfawcett, 2020-09-06:
Hi Bane, 

Wombat here.

I'm using DIYLC for the first time in awhile, and really impressed by all the amazing improvements!  One issue that bit me right away however was a problem with the spacing when snapping items to grid.

To replicate is simple, no log file required -- make new project, set grid spacing to 1 mm and try placing several eyelet/turrets in a straight row.  First one is at 0,0, but going right, next X co-ord is at 1.02, then 2,03, 3.05, 3.94, etc.  instead of just integer, 0, 1, 2, 3, 4 etc.  Y-axis suffers from the same issue of course.  It's kind of a show-stopper unfortunately.   It sounds as though this may be related to the issue identified in #552, although the context is different.

Cheers,

Paul

bancika, 2020-09-06:
Hey man, good to "see" you again! Hoping that everything is good over there.
I'll investigate. Without looking at the code, I can assume it has to do with rounding to the nearest pixel.

pmfawcett, 2020-09-07:
So some additional information.  There are three elements to this.

First, even the object positioned at 0, 0 is not (usually) displayed as centered on the 0,0 grid lines.  Here is an example:

[image]

pmfawcett, 2020-09-07:
Next, *even at the same level of magnification*, this offset from the origin varies depending on where the nominal origin is positioned relative to the larger screen.  Here is an example of where I have just moved the scroll bars, compare the position to that in the first image: 
[image]

pmfawcett, 2020-09-07:
And a third observation.... it turns out that you can have TWO separate positions that both claim to be 0,0 (_i.e._  without changing the position of the scroll bars / altering the position of the origin, just moving the part and observing where it says it is hovering).  Here's what happens when I move the mouse up one grid notch from the last image -- it STILL claims that I am hovering at 0,0!! Note that nothing else besides the hovering eyelet has been moved. 
[image]

bancika, 2020-09-10:
Hey Paul. Below is the preview of the next version. It includes many improvements re: precision.
Can you please give it a try? It seems to be solving most if not all the issues you reported.

Cheers

[diylc-4.9.0.zip](https://github.com/bancika/diy-layout-creator/files/5200992/diylc-4.9.0.zip)

pmfawcett, 2020-09-10:
Hi Bane,

I took a quick test drive (not a comprehensive test by any means), and it seems like the issue with non-integer numbers for the spacing is addressed.  However, there are couple of remaining oddities.  With the 1 mm grid, the bulls-eye of parts is still not located dead centre with respect to the grid.  While the offset is mostly consistent, it seems like maybe it can vary a bit from selection to selection?  Also, there is an oddity where certain parts (e.g. resistor) can end up drawn slightly crooked even though the end points are in the same plane.. it's like the internal points are being aligned to the grid ever so slightly differently.  See image here with subtly crooked resistor and with bulls-eye not quite on centre.  Perhaps this is evidence of a rounding bug still lurking somewhere? 

[image]

bancika, 2020-09-10:
The core of the app is now switched to double precision numbers instead of integers. But within individual components there could still be some rounding that I need to track down. I found the reason that leaded components can get crooked and fixed.
Do you have an example for other issues?
Thanks!

pmfawcett, 2020-09-10:
Aside from the crooked parts, the only issue I'm aware of with my limited testing is the issue where the bulls-eyes are not properly aligned to the grid.  Take a look at this new image -- you can see that the ends of the leads themselves actually seem to be correctly positioned, it is just the bulls-eye that is wrong (and the crookedness seems to have resulted from using the bulls-eye position rather than the lead position, as you've already discovered).  However, other parts without leads do seem to get centered incorrectly (try connector dot or solder pad).  
[image]

bancika, 2020-09-14:
I think I fixed that and many more issues I found along the way. Here's a new candidate

[diylc-4.9.0.zip](https://github.com/bancika/diy-layout-creator/files/5217348/diylc-4.9.0.zip)

bancika, 2020-11-19:
Closing for now

closed 2020-11-19 by bancika