bancika/diy-layout-creator #576: Must JDK 1.6 on Win10？

bh1rio, 2020-09-07:
Is it Must JDK 1.6 on Win10？
[image]

there is JDK 11 on my Win10......
[image]

bancika, 2020-09-07:
you could try to run it directly from the console using java 11, the command should be something like this

java -jar diylc.jar -Xms512m -Xmx1024m -Dorg.diylc.scriptRun=true -Dfile.encoding=UTF-8

bh1rio, 2020-09-07:
there is a messagebox behind the welcome.
[image]
console like this
[image]

htor, 2020-09-16:
this has probably been discussed before, but why not ship a specific JRE with the application, so the required java version is always controlled by the app maintainer and not the users? this way the JRE shipped is the one used to run DIYLC and the user don't even need the install any JRE.  

i'm no active java developer but i heard that bundling the JDK/JRE into the app distributables could perhaps solve these version problems?

i may be wrong, though!

htor, 2020-09-17:
here is some more information if you would go down that route:
* https://docs.oracle.com/javase/7/docs/technotes/guides/jweb/packagingAppsForMac.html#bundle_jre (mac)
* https://docs.oracle.com/javase/8/docs/technotes/guides/deploy/self-contained-packaging.html

bancika, 2020-12-02:
Figured out a way to include JRE in windows installation as well, closing.

closed 2020-12-02 by bancika