bancika/diy-layout-creator #577: [Mac OS] Command key shortcuts broken, working only intermittently

iiimonfiiire, 2020-09-10:
Running DIYLC newest version 4.8.0 on Mac OS (10.14.6)

**Description:**
Keyboard shortcuts using Command key are broken - sometimes they work, sometimes they do not. Not sure what specific reproduction steps need to be taken, as it happened to me quite often after simply working with a .DIY file. Commands that were working/not working intermittently:
Copy [Command + C]
Paste [Command + V]
Edit component settings [Command + E]

[log-2020-09-10 10.01.42.txt](https://github.com/bancika/diy-layout-creator/files/5200040/log-2020-09-10.10.01.42.txt)

bancika, 2020-09-16:
Is it the same when you switch from the component tree to the toolbar (from Config menu)?

bancika, 2020-12-11:
assuming it was fixed with the new component tree.  Please comment if still an issue

closed 2020-12-11 by bancika