bancika/diy-layout-creator #580: Black screen with Windows 10

chemical-oli, 2020-09-13:
[log-2020-09-14 12.04.37.txt](https://github.com/bancika/diy-layout-creator/files/5215317/log-2020-09-14.12.04.37.txt)
Hi im having problems with Windows 10, the main screen is black and the "mid-scroll" arrow appears in menu items and across the black screen.  Appears not to be rendering properly.
[image]

Using Java runtime environment 1.8.0_261

Link to [screenshot](https://imgur.com/a/aqliFou) just in case

bancika, 2020-09-14:
Please try toggling "Hardware Acceleration" setting from the Config menu.

chemical-oli, 2020-09-17:
Thanks this seems to have fixed the problem.

closed 2020-09-17 by chemical-oli

bancika, 2020-09-17:
Cool! Seems like it's not compatible with all graphic cards or maybe drivers.