bancika/diy-layout-creator #582: Zoom and moving through sheet

organbender, 2020-09-20:
Hi,
I just updated to diylc 4.9.0. I write to make a suggestion for you next version. 

Overall the program is cool, yet a bit stubborn at times. But the thing that bothers me the most is zooming in/out and moving through the layout worksheet (macbook with trackpad). The zoom is quite not natural, it seems to scale randomly (actually this worked better on previous version I had; 3.57.0). 
As for moving through the sheet, let's say it is just time consuming to aim to the side or bottom cursors. The arrows are ok for small scale movements but useless to go longer distances. I use many programs for different purposes and the best ones are usually the ones that provide the best workflow. 

I use a lot of both of these features when designing layouts. These would make so much more sense (at least for macbook users) to go with the common ways : navigate by scrolling with 2 fingers on the trackpad in any direction; and zooming using either "command+(up/down) arrow", "command+ +/-", or by pinching on the trackpad. 

Thanks for taking that into consideration
I look forward to work with your next improvements

bancika, 2020-09-21:
Hi there.
Thanks for the suggestion.  What do you mean by scaling randomly? There are few pre-defined zoom levels and it always goes up or down step by step. Is that what you are seeing? Also I don't remember changing anything there for a while. What is different compared to the older version?

As far as th  two-finger trackad operation...someone asked for it in #501 but there are two main reasons why it wasn't done so far
1) I don't have a mac and nobody I know has one, so it's impossible for me to test while developing. I would have to code blindly and ask people to help testing, then fix issues blindly and so on.
2) everything so far in DIYLC is multi-platform, meaning that it works the same across all OSs. This would be specific development for Mac users which make up for ~20% of the user base, so I usually prioritize items that can benefit all users. I have very little free time to work on the project, so I have to prioritize somehow.
Cheers

organbender, 2020-09-21:
Hi,

I can understand you work on it in your spare time and it is mostly pc
users.

About the zooming, I should have been more specific. I acts "as if" it was
random. It is erratic, not stable. If I zoom with "command+trackpad", it
zooms but somehow tends to jump back to a bigger scale before I release the
command button. So it is not possible to take it to a specific scale
increment this way, if not by luck. I can use the scale dial in the down
right corner though.

Now for the movement, it appears I was wrong. I just opened the program and
it does move with the trackpad like I wished. Perhaps I was distracted and
was in the older version at that moment. Sorry for that.

If you need someone to test some stuff on mac, let me know. I can spare
some time to contribute.

Cheers

htor, 2020-09-22:
i just want to say that the latest version (4.9.0) solves a big mac problem, like the previously unusable Searchable Tree, now fully functional and fast to use. i think a lot of users had problems with the mac app because of that. we'll see. thank you for that @bancika.

this issue is really just #501, so maybe this discussion should be moved there.

organbender, 2020-09-22:
All right, I will go check #501. It was nice talking to you. And thanks for
that cool program. It makes building electronic more enjoyable.

François

bancika, 2020-09-22:
Cool, closing this one for now.

closed 2020-09-22 by bancika