bancika/diy-layout-creator #584: Cannot select/delete elipses

JRabbit54, 2020-09-25:
When trying to make a small circle, if it collapses into a line, it becomes a permanent mark on your layout. It cannot be selected, therefore it cannot be deleted. Curiously, when I "select all" it gets selected along with everything else, but otherwise it is as though the application has lost track of the object.

bancika, 2020-10-05:
you can select it from the context menu in the component tree on the left. Ellipse -> Select All or something

closed 2020-10-05 by bancika

JRabbit54, 2020-10-05:
Yes, you can "Select All" from the component tree, which of course selects all elipses on the layout. I only want to delete the ones that are messed up. There is no way that I can find to only delete the ones that collapsed into a line and became useless eyesores.