bancika/diy-layout-creator #585: File opening fails because of issue with trimmer potentiometer (array element type mismatch).

pmfawcett, 2020-09-25:
Hi Bane,

Here's a DIY file (zipped) that won't open with 4.9.0.  I'm not sure if it's because I created it to test the 4.9.0 candidate, or if there is something else going on (the candidate executable doesn't seem to work either).  I can't open it with 4.8.0 either as it's from a newer version.

Log (below) seems to suggest that there is an array element type mismatch associated with a trimmer potentiometer.  I think I might have tried putting flexible leads on it? I don't recall for sure.   Anyway, I hope the work is not lost.

Cheers,

Wombat

[CEDist_JTM45.zip](https://github.com/bancika/diy-layout-creator/files/5285607/CEDist_JTM45.zip)

18:03:53,027  INFO [main] ConfigurationManager - Initializing configuration
18:03:53,345  INFO [main] ConfigurationManager - Saving configuration
18:03:53,410  INFO [main] DIYLCStarter - Loading languages...
18:03:53,411  INFO [main] LangUtil - Loading translation for English
18:03:53,483 DEBUG [main] DIYLCStarter - Java version: 1.8.0_101-b13 by Oracle Corporation
18:03:53,484 DEBUG [main] DIYLCStarter - OS: Windows 7 6.1
18:03:53,484  INFO [main] DIYLCStarter - Starting DIYLC with working directory C:\Program Files (x86)\DIYLC
18:03:53,484  INFO [main] DIYLCStarter - Configuration dump start.
18:03:53,485  INFO [main] DIYLCStarter - antiAliasing = true
18:03:53,485  INFO [main] DIYLCStarter - hiQualityRender = true
18:03:53,485  INFO [main] DIYLCStarter - stickyPoints = true
18:03:53,485  INFO [main] DIYLCStarter - metric = true
18:03:53,485  INFO [main] DIYLCStarter - snapTo = Components
18:03:53,485  INFO [main] DIYLCStarter - continuousCreation = false
18:03:53,485  INFO [main] DIYLCStarter - wheelZoom = true
18:03:53,486  INFO [main] DIYLCStarter - outline = false
18:03:53,486  INFO [main] DIYLCStarter - showRulers = true
18:03:53,486  INFO [main] DIYLCStarter - showGrid = true
18:03:53,486  INFO [main] DIYLCStarter - highlightContinuityArea = false
18:03:53,486  INFO [main] DIYLCStarter - defaultTemplatesImported = true
18:03:53,486  INFO [main] DIYLCStarter - defaultBlocksImported = true
18:03:53,487  INFO [main] DIYLCStarter - Configuration dump end.
18:03:53,528  INFO [main] DIYLCStarter - Creating the main frame...
18:03:53,528 DEBUG [Thread-3] DIYLCStarter - Starting font pre-loading
18:03:53,530 DEBUG [Thread-3] ResourceLoader - Looking for files in: fonts
18:03:53,530  INFO [Thread-3] DIYLCStarter - Dynamically loading font: consola.ttf
18:03:54,085  INFO [Thread-3] DIYLCStarter - Dynamically loading font: lucon.ttf
18:03:54,085  INFO [Thread-3] DIYLCStarter - Dynamically loading font: square721_bt.ttf
18:03:54,088  INFO [main] Presenter - Current DIYLC version: 4.9.0
18:03:54,101  INFO [main] Presenter - Checking if variants need to be updated
18:03:54,101  INFO [main] Presenter - Loading component types.
18:03:54,102 DEBUG [main] Utils - Scanning jar: /C:/Program Files (x86)/DIYLC/lib/diylc-core.jar
18:03:54,132 DEBUG [main] Utils - Scanning jar: /C:/Program Files (x86)/DIYLC/library/main.jar
18:03:54,312 DEBUG [main] JarScanner - Found JAR file: main.jar
18:03:54,312 DEBUG [main] JarScanner - Scanning main.jar
18:03:54,368  INFO [main] ClassLoaderUtil - Adding file to the classpath: C:\Program Files (x86)\DIYLC\library\main.jar
18:03:54,371 DEBUG [main] JarScanner - Loaded class: org.diylc.components.Abstract3LegSymbol
18:03:54,371 DEBUG [main] JarScanner - Loaded class: org.diylc.components.AbstractLeadedComponent
18:03:54,371 DEBUG [main] JarScanner - Loaded class: org.diylc.components.AbstractMultiPartComponent
18:03:54,371 DEBUG [main] JarScanner - Loaded class: org.diylc.components.AbstractRadialComponent
18:03:54,371 DEBUG [main] JarScanner - Loaded class: org.diylc.components.AbstractSchematicLeadedSymbol
18:03:54,371 DEBUG [main] JarScanner - Loaded class: org.diylc.components.boards.AbstractBoard
18:03:54,371 DEBUG [main] JarScanner - Loaded class: org.diylc.components.boards.BlankBoard
18:03:54,371 DEBUG [main] JarScanner - Loaded class: org.diylc.components.boards.Breadboard
18:03:54,371 DEBUG [main] JarScanner - Loaded class: org.diylc.components.boards.EyeletBoard
18:03:54,372 DEBUG [main] JarScanner - Loaded class: org.diylc.components.boards.MarshallPerfBoard
18:03:54,372 DEBUG [main] JarScanner - Loaded class: org.diylc.components.boards.PerfBoard
18:03:54,372 DEBUG [main] JarScanner - Loaded class: org.diylc.components.boards.ProtoBoard
18:03:54,372 DEBUG [main] JarScanner - Loaded class: org.diylc.components.boards.TagStrip
18:03:54,372 DEBUG [main] JarScanner - Loaded class: org.diylc.components.boards.TerminalStrip
18:03:54,372 DEBUG [main] JarScanner - Loaded class: org.diylc.components.boards.TriPadBoard
18:03:54,372 DEBUG [main] JarScanner - Loaded class: org.diylc.components.boards.VeroBoard
18:03:54,372 DEBUG [main] JarScanner - Loaded class: org.diylc.components.chassis.ChassisPanel
18:03:54,372 DEBUG [main] JarScanner - Loaded class: org.diylc.components.chassis.EllipticalCutout
18:03:54,373 DEBUG [main] JarScanner - Loaded class: org.diylc.components.chassis.PolygonalCutout
18:03:54,373 DEBUG [main] JarScanner - Loaded class: org.diylc.components.chassis.RectangularCutout
18:03:54,373 DEBUG [main] JarScanner - Loaded class: org.diylc.components.connectivity.CopperTrace
18:03:54,373 DEBUG [main] JarScanner - Loaded class: org.diylc.components.connectivity.CurvedTrace
[... 417 lines cut ...]
	at com.thoughtworks.xstream.mapper.MapperWrapper.realClass(MapperWrapper.java:30)
	at com.thoughtworks.xstream.mapper.MapperWrapper.realClass(MapperWrapper.java:30)
	at com.thoughtworks.xstream.mapper.MapperWrapper.realClass(MapperWrapper.java:30)
	at com.thoughtworks.xstream.mapper.MapperWrapper.realClass(MapperWrapper.java:30)
	at com.thoughtworks.xstream.mapper.MapperWrapper.realClass(MapperWrapper.java:30)
	at com.thoughtworks.xstream.mapper.CachingMapper.realClass(CachingMapper.java:45)
	at com.thoughtworks.xstream.core.util.HierarchicalStreams.readClassType(HierarchicalStreams.java:29)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.start(TreeUnmarshaller.java:133)
	at com.thoughtworks.xstream.core.AbstractTreeMarshallingStrategy.unmarshal(AbstractTreeMarshallingStrategy.java:32)
	at com.thoughtworks.xstream.XStream.unmarshal(XStream.java:1058)
	at com.thoughtworks.xstream.XStream.unmarshal(XStream.java:1042)
	at com.thoughtworks.xstream.XStream.fromXML(XStream.java:922)
	at org.diylc.serialization.ProjectFileManager.parseV3File(Unknown Source)
	at org.diylc.serialization.ProjectFileManager.deserializeProjectFromFile(Unknown Source)
	at org.diylc.presenter.Presenter.loadProjectFromFile(Unknown Source)
	at org.diylc.swing.ActionFactory$OpenAction$1.doInBackground(Unknown Source)
	at org.diylc.swing.ActionFactory$OpenAction$1.doInBackground(Unknown Source)
	at org.diylc.swing.gui.MainFrame$6.doInBackground(Unknown Source)
	at javax.swing.SwingWorker$1.call(Unknown Source)
	at java.util.concurrent.FutureTask.run(Unknown Source)
	at javax.swing.SwingWorker.run(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.lang.Thread.run(Unknown Source)
18:06:00,843  INFO [AWT-EventQueue-0] MainFrame - Closing the app

pmfawcett, 2020-09-26:
Deleting the potentiometer from the XML file allows the file to successfully open

pmfawcett, 2020-10-05:
I don't really have a way to tell, as I can't recall the exact steps that were used to create the potentiometer.  But here is the block of XML that caused the problem.. Cutting this out allows it to open in 4.9. 

<diylc.passive.TrimmerPotentiometer>
      <name>VR1</name>
      <alpha>127</alpha>
      <controlPoints class="java.awt.Point-array">
        <java.awt.Point x="8.42" y="2.61"/>
        <java.awt.Point x="8.62" y="2.11"/>
        <java.awt.Point x="8.82" y="2.61"/>
      </controlPoints>
      <orientation>_270</orientation>
      <taper>LIN</taper>
      <bodyColor hex="ffffe0"/>
      <borderColor hex="8e8e38"/>
      <display>NAME</display>
      <type>FLAT_XLARGE</type>
    </diylc.passive.TrimmerPotentiometer>

pmfawcett, 2020-10-05:
I notice that if I create a new potentiometer, it is no longer using java.awt.Point-array, just point, so hopefully that explains it.  e.g.:

diylc.passive.TrimmerPotentiometer>
      name>VR1/name>
      alpha>127/alpha>
      controlPoints>
        point x="3.675700570057006" y="2.145532553255326"/>
        point x="3.875700570057006" y="1.6455325532553262"/>
        point x="4.075700570057006" y="2.145532553255326"/>
      /controlPoints>
      orientation>_270/orientation>
      taper>LIN/taper>
      bodyColor hex="ffffe0"/>
      borderColor hex="8e8e38"/>
      display>NAME/display>
      type>FLAT_XLARGE/type>
    /diylc.passive.TrimmerPotentiometer>

bancika, 2020-10-05:
Ok, I confirmed that it's an issue that affects only files created with the
candidate, so I don't think we need to fix anything.

As a test, I made a file with 4.8.0 version that contains trimmers and it
opens fine in 4.9.0.

Candidate version had issues and those were fixed before releasing the
4.9.0.

Cheers

closed 2020-12-02 by bancika