bancika/diy-layout-creator #586: Zoom with trackpad does not work correctly (PC, version 4.9.0)

pmfawcett, 2020-09-26:
I saw a recently closed issue regarding the Mac version on this, but I also am having trouble with zoom using a PC laptop.  
The two-finger gesture makes it super-rapidly increase the zoom level (it goes from 50% to 300% very quickly - it's way too twitchy to be useful).  But it is a one-way trip.... Once it hits 300%, it cannot be reduced (actually it will alternate between showing 250% and 300% but it always lands on 300%).  With "Mouse wheel zoom" off, you need to hit CTRL and do a two-finger to get the same effect, but it likewise is a one-way trip -- you need to use the percentage setting on the bar to get it small again.

pmfawcett, 2020-09-26:
I see I was using the wrong paradigm for zoom -- I should have been using stretch and pinch gestures.  Instead I was trying to "roll the mouse wheel" up or down.  I agree with the mac user's comment that it is quite annoying on a laptop to try and move the canvas around.  There should be a key or gesture that pops up a "hand" for dragging the canvas about rather than having to use the scroll bars or arrows, which is quite cumbersome.

bancika, 2020-12-13:
most, if not all is fixed in 4.12, pending release

closed 2020-12-13 by bancika