bancika/diy-layout-creator #587: Feature request: Bottom view

ngmacha, 2020-09-27:
Hello - Thank you for a great software! There is one feature that would make it even better: the "flip-over" view, showing the board from below, similar to KiCAD's "Flip Board View". This will be very useful to easily identify where to cut the traces on the stripboards. Hope it is not too difficult to implement? I'm no Java programmer, but gladly help whatever I can.
Thank you - Mac Ha

bancika, 2020-10-05:
I understand but it goes against the basic idea of DIYLC. One way to do it is to draw two boards side by side and have one represent top and the other bottom view. That's more in line with how DIYLC works.

ngmacha, 2020-10-06:
Hello, thank you for the response. I don't quite understand why the addition of a flip-view goes against the basic idea of DIYLC - it's just another view, possibly only for printout? But anyway let's close this issue in meantime. Thanks again for a good tool.

closed 2020-10-06 by ngmacha

bancika, 2020-10-06:
Here's why. DIYLC is a WYSIWYG drawing tool, kind of like Corel Draw. When
you flip the board, some of the components would need to disappear, others
need to appear. Also traces for PCBs. And if you want to print or export
such a layout, you can only do one side at a time.